pvvx/ATC_MiThermometer #354: Asking for passcode

ezerw, 2023-06-12:
Hi, 
First of all thanks for these amazing tools. 
I've an issue with my *Xiaomi Mijia (LYWSD03MMC)*. I've flashed the custom firmware v4.3 (and didn't change anything in the config) and now when I try to connect via bluetooth to it it ask for a passcode.

Is there a default passcode that can I use to unlock it and put the original firmware back?
If not, should I connect it to USB to be able to use [hardware flasher](https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html) and do it through cable?

pvvx, 2023-06-13:
The PIN code can be reset only by clearing a special area in Flash. Or the whole Flash and then writing a new firmware.

ezerw, 2023-06-15:
Thanks @pvvx Problem is now I can't get it to connect (because of the passcode) to do this. Is there a way to remove the passcode (even if it means soldering, etc)?

pvvx, 2023-06-16:
I meant https://pvvx .github.io/ATC_MiThermometer/USBCOMFlashTx.html
1. Erase All Flash
2. Write to Flash

gregrow, 2024-02-03:
The ease of recovery is super cool, but... well, much of insecure. Anyone within BT/BLE range can alter my devices! Maybe this is just a thermometer, so we'd rather take it easy, don't we? ;) IMHO in some setups - like mine - such attacks may cause significant financial (and ecological!) losses. I use Home Assistant based automation with a number of LYWSD03MMC sensors to control my house heating system. If sensor readings get false (or even if unavailable) the automation may drive unnecessary high energy consumption. Therefore, I'm looking for ways to harden the security of the system. Perhaps I do not know all the features and perhaps I overlooked the security means available in the firmware. Let us know what you think about it. Is it possible at all to make the thing secure given the hardware limitations?
Victor, sincere gratitude to you for developing and maintaining this nice piece of software. 
Cheers!

pvvx, 2024-02-04:
1. The PIN code protects against changes to the thermometer configuration.
2. Using Bindkey with BLE ad encryption protects against substitution.