pvvx/ATC_MiThermometer #355: negative number display problem

oukene, 2023-06-20:
https://esphome.io/components/display/pvvx_mithermometer.html

---------------------------------------------------------------------------------
display:
  -platform: pvvx_mithermometer
    # ...
    lambda: |-
      // Print -2.1 as big number (first row)
      it. print_bignum(-2.1);
      // Print °C next to the big number
      it.print_unit(pvvx_mithermometer::UNIT_DEG_C);
      // Print 88 as small number (second row)
      it. print_smallnum(88);
      // Print % next to the small number
      it. print_percent(true);
      // Print the low battery symbol
      it. print_battery(true);

      // Print a happy smiley. Results in " ^_^ "
      it. print_happy(true);
      // Print a sad smiley. Results in " -∧- "
      it. print_sad(true);
      // The comination of happy and sad simley results in " ΔΔΔ "

      // Print round brackets around the simley
      it. print_bracket(true);
      // The final result is "(ΔΔΔ)"
----------------------------------------------------------------------------------

I wrote it the same as the sample in the link above, but negative numbers are not displayed.

0.0 and 0 are printed.

I tried using ATC_v42.bin, ATC_v43.bin for the firmware version, and the esphome version is 2023.5.5.
What should I check?

https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html

After accessing the page above, if a negative value is output randomly, it will be output normally.

When I use display in esphome, it shows 0.0, 0. And the positive numbers are displayed well.

pvvx, 2023-06-21:
?

[TelinkMiFlasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html):

[image]

[image]

oukene, 2023-06-21:
After some more checking yesterday, it turned out that the problem was with esphome pvvx_mithermometer component.

closed 2023-06-21 by oukene