pvvx/ATC_MiThermometer #357: New chart style in GraphMemo?

lukaszswitaj, 2023-06-26:
Hi,

There is a screenshot of a pretty nice set of charts in Readme.md:

[image]

Is this some new way to visualize downloaded sensor data directly in the browser? The graphs drawn here look completely different: https://pvvx.github.io/ATC_MiThermometer/GraphMemo.html

lukaszswitaj, 2023-06-27:
Forgive me, this illustration is just below the link to [GraphMemo.html](https://pvvx.github.io/ATC_MiThermometer/GraphMemo.html). It's very confusing.

By the way, maybe it's worth posting an instruction on how to convert timestamp from generated CSV files to date-time format in Calc and Excel.

pvvx, 2023-06-27:
The csv file describes the conversion formula in the first row of the first column:
`Time-Data=(A2/86400)+25569`

closed 2023-06-28 by lukaszswitaj

evlo, 2023-09-27:
I don't understand the excel? formula or in general _time-data?_ what does that actually mean

[image]

evlo, 2023-09-27:
I do not understand

this is what I get on the device so date and time is correct

[image]

these are results in excel + formula I'm using

[image]

`rrrr-mm-dd h:mm;@` is cell froamt, but I do have czech localisation of excel, sadly

pvvx, 2023-09-27:
After the firmware, you need to set the clock in the thermometer - "Set time".
[Instability of the time count on the thermometer.](https://github.com/pvvx/ATC_MiThermometer/issues/56)
There are already several applications for synchronizing the time on the thermometer.
Example:
**ESPHome**: [synchronize the time once a day](https://esphome.io/components/display/pvvx_mithermometer.html#only-synchronize-the-time-once-a-day)