pvvx/ATC_MiThermometer #358: [Feature request] Please make log averaging optional

joshuakraemer, 2023-06-27:
At the moment, the log data is always stored as the average of n measurements. Instead, I would like to simply store every n'th measurement without averaging. With averaging, it's impossible to compare/combine logged and published measurements. It's also impossible to tell if a temperature was above or below a given limit at a certain timepoint. Could you please introduce an option to enable/disable averaging?

pvvx, 2023-06-27:
1. The available flash volume allows you to save only 19632 measurements. 
2. Additional battery current is required for each recording.
3. Additional battery current is also required to read and transmit values. Frequent readings of all records severely drain the battery.  On a good BT adapter, the process of reading all stored values takes seconds, and on terrible ESP32 - minutes.
Getting 50 measurements on ESP32 (12 sec):
[image]
(current on the thermometer mA/ms):
BT adapter or Smartphone (1.2..1.5 sec):
[image]
(mA/ms scale)

joshuakraemer, 2023-06-28:
Thanks for your answer. I think I didn't explain my request correctly. I don't want to save all measurements. In case of a 30 min logging interval, at the moment all measurements inside the logging interval are averaged. Instead, I want just the last of those measurements to be saved. The amount of data saved would be the same (1 measurement every 30 minutes), but the saved data would represent the actual measurement at the given timepoint instead of an average.

pvvx, 2023-06-28:
Then this goal is lost:

> It's also impossible to tell if a temperature was above or below a given limit at a certain timepoint. 

As a result, there are many options, such as recording the average, minimum and maximum for the period.
All this variety of options is easier to implement in external software, rather than loading a weak chip and battery.
The main use case of thermometers is designed to work in the "smart home" system with constant collection and processing of the received measurements. It is possible and much easier to implement any logging options there.

joshuakraemer, 2023-06-28:
Thanks, I understand all of those points. With more values averaged, you get higher accuracy but also higher lag. The choice of averaging is thus a tradeoff. With log averaging, the lag is too high for my use case. This is why I would prefer lower lag, accepting the loss in accuracy.

I've already implemented logging on a central server. The internal memory of the thermometers serves as a backup only. In case of a lost connection, I would like to read the logged values from the thermometer and add them to the central server database to fill the gaps. But because of the additional averaging in the thermometer's log, the data don't match in case of a change in termperature (too much lag in the thermometer's log).

pvvx, 2023-06-28:
> This is why I would prefer lower lag, accepting the loss in accuracy.

> In case of a lost connection, I would like to read the logged values from the thermometer and add them to the central server database to fill the gaps.

?

[image]

pvvx, 2023-06-28:
For clarity, I describe the assembly of the version with the ability to [measure 2 analog channels ](https://github.com/pvvx/ATC_MiThermometer/blob/master/src/app_config.h#L47) instead of the TH sensor. 
```
/* Special DIY version - Voltage Logger:
 * Temperature 0..36.00 = ADC pin PB7 input 0..3.6V, LYWSD03MMC pcb mark "B1"
 * Humidity 0..36.00 = ADC pin PC4 input 0..3.6V, LYWSD03MMC pcb mark "P9"
 * Set DIY_ADC_TO_TH 1 */
#define DIY_ADC_TO_TH 	1
```
Settings:

[image]

We feed the sine from the generator:

[image]

Triangular signal from the generator:

[image]

Similarly, with a temperature and humidity sensor, measurements can be set in increments of 63 ms.

pvvx, 2023-06-28:
>  In case of a lost connection

To get the values from the thermometer, "BLE connection" is not used. In the latest firmware versions, it was necessary to limit the "connection interval" for the stability of the connection with various BT adapters. As a result, when connected, consumption increases. The connection to the thermometer is used in cases of changing settings and rare readings of the log (such as once every 3 months or when changing the battery).

joshuakraemer, 2023-06-28:
> To get the values from the thermometer, "BLE connection" is not used.

I'm sorry, I've used the wrong terminology. With "lost connection", I've meant a situation where broadcast (advertising) messages are no longer received and logged by the server (for example, in case of a power failure of the server), but the thermometer keeps working and can still save measurements on its internal flash memory.

> No averaging and a complete log.

Unfortunately, a 1x logging interval doesn't work for me, because the flash memory will be full too soon.

> Sampling one of several values is by no means a correct measurement.

I disagree. Averaging is not better than sampling. Those are just two different methods with different tradeoffs, which are suitable for different use cases. In my case, averaging of time series data over several minutes is not correct. The following plot shows a test with temperature changing from 25°C to 0°C and back to 25°C. Blue dots represent broadcast data (1 min measurement interval) and red dots represent data saved on the internal memory (5 min logging interval). As you can see, because of the averaging, the data doesn't match.

[image]

I would really appreciate an option to enable/disable averaging, so the right logging method can be chosen according to the use case.

joshuakraemer, 2023-06-29:
Thank you very mich for your thoughts.

> Here it is necessary to apply a correction - subtract the offset of the averaging window and the graphs will match.

This would just help to create a nice looking chart. But it still wouldn't be correct to combine data processed in different ways (broadcast data without averaging and logged data with averaging) and further process/analyze the combined data.