pvvx/ATC_MiThermometer #359: [Feature request] Adding a SHT3x external temperature sensor

jmbinette, 2023-06-28:
Hi,

I know this may have been discussed before but the suggestion of adding 1-wire sensors (DS18B20) was, in my opinion, too far away from the original project.

I would like to know if it would be possible to integrate the capability to replace or even add a 2nd temperature sensor using a SHT3x i2c sensor (or any other i2c sensor)

I tried to wire SDA/SCL on a rev B1.9 hardware LYWSD03MMC using firmware 4.3 but I end-up with a -45C reading. Probably because the SHT31 (waterproof version) I was using was not the the same i2c address.

Would that be possible ? I think this would be a great addition because other solutions involving ESP controller are not suitable for battery operated sensors. The Xiaomi sensor is the best platform in my opinion.

A reed sensors can be integrated, an external contact as well ... an external i2c device would be a great addition.

landrysik, 2023-07-24:
You should replace the sensor with the same sensor - LYWSD03MMC has SHT40 sensor inside (and one revision has SHTC3) NOT SHT31.
I successfully replaced default sensors them with external ones on a wire. Photos and info here (just toss this in google translate, original language is Czech)  https://www.facebook.com/groups/2232679967058877/permalink/3469388833387978/

jmbinette, 2023-07-25:
The temperature sensor must be immerged in water. I don't think there is a waterproof SHT40 ?

landrysik, 2023-07-26:
Tee SHT3X is as (non)waterproof as SHT40. The important aspect is the enclosure. I needed waterproof version aswell - Contact this seller with request for fully waterproof version. By default there is a hole with membrane so the sensor can read humidity.
https://www.aliexpress.com/item/1005003916357191.html

jmbinette, 2023-07-26:
Thanks for that ! Will contact them.

Do you think there would be a possibility to add 2 sensors ? Maybe modifying the firmware ?
The ideal would be : 1 external temperature (waterproof) + 1 ambient sensor (Temp + humidity)

Thanks !

landrysik, 2023-07-27:
That is mainly a question for @pvvx but I would say this is out of the scope for this project and would only complicate things (what value to show on LCD, what value to transmit over BT...) Considering the price of those thermometers - just get one for ambient readings and second one for remote readings....

jmbinette, 2023-08-09:
> That is mainly a question for @pvvx but I would say this is out of the scope for this project and would only complicate things (what value to show on LCD, what value to transmit over BT...) [...]

Thanks ! I received my waterproofed version of SHT40 yesterday and it worked perfectly !

[image]

landrysik, 2023-08-09:
You're welcome! Consider adding capacitor(s) to unpopulated pads (C24 or C25). I will improve battery life and operation during lower battery levels where the battery has increased internal resistance and for example during configure command over bluetooth it fails due to drop in voltage. I added 100uF but any value you have lying around helps

pvvx, 2023-10-23:
The firmware uses an increased I2C bus frequency to reduce consumption due to the sensor polling speed. The datasheet on the sensor allows you to use an even higher speed ...
But, when using an external sensor on the wires, you need to lower the frequency of the I2C bus.
https://github.com/pvvx/ATC_MiThermometer/blob/master/src/i2c.c#L15