pvvx/ATC_MiThermometer #362: TypeError: getDevVersion(...).then is not a function issue

Baptiste-mrch, 2023-07-13:
Hello,

I'm trying to give a name to my Xiaomi lywsd03mmc sensor. I flashed the firmware using the https://github.com/atc1441/ATC_MiThermometer tool, and see that I have to use this tool in order to give my sensors a name.
Whatever I tried, I still faced this issue:
[image]

Did someone know why I cannot connect to my sensors using this tool, but the original ATC_MiThermometer tool works fine?

Thanks!

pvvx, 2023-07-16:
> I flashed the firmware using the https://github.com/atc1441/ATC_MiThermometer tool

This is a different version and works with [atc1441 TelinkFlasher](https://atc1441.github.io/TelinkFlasher.html)

The [firmware from atc1441](https://github.com/atc1441/ATC_MiThermometer) does not have a name replacement function.
To replace the firmware with the version from this repository, you will need to use [atc1441-TelinkFlasher](https://atc1441.github.io/TelinkFlasher.html) and [ATC_v43.bin](https://github.com/pvvx/ATC_MiThermometer/raw/master/ATC_v43.bin) file.

Baptiste-mrch, 2023-07-17:
> > I flashed the firmware using the https://github.com/atc1441/ATC_MiThermometer tool
> 
> 
> 
> This is a different version and works with [atc1441 TelinkFlasher](https://atc1441.github.io/TelinkFlasher.html)
> 
> 
> 
> The [firmware from atc1441](https://github.com/atc1441/ATC_MiThermometer) does not have a name replacement function.
> 
> To replace the firmware with the version from this repository, you will need to use [atc1441-TelinkFlasher](https://atc1441.github.io/TelinkFlasher.html) and [ATC_v43.bin](https://github.com/pvvx/ATC_MiThermometer/raw/master/ATC_v43.bin) file.
> 
> 
> 
> With the [ATC_v43.bin](https://github.com/pvvx/ATC_MiThermometer/raw/master/ATC_v43.bin)  will work [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html).

Works great now, thanks!

closed 2023-07-17 by Baptiste-mrch