pvvx/ATC_MiThermometer #367: Slow sensor updates, cant figure out

minermartijn, 2023-08-18:
Soo, first of all, your firmware is awesome. And i've upgraded 2 of my ATC LYWSD03MMC to use yours.
And it all works, only thing that i cant seem to figure out is that sometimes it updates within a minute on HA.
And sometimes it takes over 15 min, sometimes it becomes unavailable..

I got one very close to my bluetooth adapter, and one a bit further away.
If you need more info i will ofcourse share it, but im at work and cant access the bluetooth firmware side of things right now.

pvvx, 2023-08-19:
The update time from the state of the original Xiaomi version does not depend on the new firmware code. This time is normalized in accordance with the original official version of the program and the quality of your device transmitting the update code.

minermartijn, 2023-08-21:
I think i explained wrong.
You are talking about updating the firmware right?

But i mean that it sometimes takes 10 minutes to update the temperature or humidity. I wish i would update like every 5 seconds and cant seem to find out how.

pvvx, 2023-08-21:
It depends on the BT adapter and the internal settings in HA.

minermartijn, 2023-08-22:
Then it must be my adapter. I uae the Rolio bluetooth 5.1 adapter.

And i have my settings on 2500ms i think. Not at home right now.

[image]

As you can see on my screenshot it sometimes takes very long to update

mmib, 2023-08-24:
have the same issue. Set interval to 8000ms and measure interval to 35 to receive updates each 4-5 minutes. At the beginning it worked but then stop (look at this gap at night)
[image]

pvvx, 2023-08-27:
[image]

Step 60 seconds:

[image]

Part of the devices, reception on one USB-BT adapter, only BLE thermometers:

[image]

[image]

minermartijn, 2023-08-28:
Aaah, you are using a different bluetooth integration, im going to try that one now, saw i could install it using hasc.

minermartijn, 2023-08-28:
Oke, quick update! 
The other integration made all the difference for me, its updating atleast within a minutes now.
And works so much better!!! Thanks PVVX! YOU THE MAN!

[image]

minermartijn, 2023-08-29:
This issue can be closed, atleast for me. Thanks again pvvx!! :)

closed 2023-08-29 by minermartijn

pvvx, 2023-08-29:
I use two of BT integrations ("[Bluetoth](https://www.home-assistant.io/integrations/bluetooth/)" and "[Passive BLE monuitor](https://github.com/custom-components/ble_monitor)") and two USB-BT adapters.