pvvx/ATC_MiThermometer #368: ClearGrass Temp & RH: Not found Telink OTA service!

sebi5361, 2023-08-26:
I have a CGG1 variant which advertises as **ClearGrass Temp & RH** when pairing through Bluetooth .
The standard firmware works fine and I even integrated the device to homeassistant using HACS - Passive BLE monitor integration.
I would like to install this alternative firmware to improve the battery lifetime.
Below are the logs I obtain when trying to connect using the [Telink Flasher for Mi Thermostat](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html).
Could you help?
```
2:59:04 PM: Searching for devices
2:59:15 PM: Connecting to: ClearGrass Temp & RH
2:59:20 PM: Not found Telink OTA service!
2:59:20 PM: Disconnected.

2:59:58 PM: Searching for devices
3:00:21 PM: Get Advertising MAC (Web Experimental Features!)
3:00:21 PM: Connecting to: ClearGrass Temp & RH
3:00:22 PM: MiVer3, DevID: 0x0347-CGG1, FnCnt: 62, CtrID: 0x3230 Not bound, Request APP to register and bind, Old version certification, Request APP to register and bind, MAC: 582D3410CD12, Capability: 0x0D - front binding
3:00:27 PM: Not found Telink OTA service!
3:00:27 PM: Disconnected.

10:27:02 AM: Searching for devices
10:27:39 AM: Get Advertising MAC (Web Experimental Features!)
10:27:39 AM: Connecting to: ClearGrass Temp & RH
10:27:39 AM: MAC: ?
10:27:42 AM: Not found Telink OTA service!
10:27:42 AM: Disconnected.
```

[image]

pvvx, 2023-08-27:
https://github.com/pvvx/ATC_MiThermometer/issues/41#issuecomment-812803456

sebi5361, 2023-08-27:
Oh OK. Thank you for the link.
In nRF connect, as the UUID I read is 0xFE59, my CGG1 is a nRF variant and there is no alternative firmware.
Too bad...

closed 2023-08-27 by sebi5361

Cr4z33, 2023-12-20:
> [#41 (comment)](https://github.com/pvvx/ATC_MiThermometer/issues/41#issuecomment-812803456)

I am in the same boat.

Although I followed the link I couldn't figure out what to do exatly.

Any help please? 😅

pvvx, 2023-12-20:
>  Connecting to: ClearGrass Temp & RH

The Qingping thermometer name has the following initial characters:: 
"Qingping Temp & RH M"
"Qingping Temp RH M"
"Qingping Temp RH Lite"