pvvx/ATC_MiThermometer #370: Sensor stop to update

mmib, 2023-08-28:
Has last FW version, 1.6B hardware revision of LYWSD03MMC
With default settings sensor do work fine for some time (1-2 day after flashing) and then start to update data just once per 6-8 hours
[image]
 Switched to https://github.com/atc1441/ATC_MiThermometer and everything works fine
[image]
What can be the reason?

pvvx, 2023-09-02:
More than 20 thermometers have been operating for the second year. Some of them are only for stability tests. Everything is always updated to the latest version.

[image]

What kind of integration and thermometer settings do you use?

mmib, 2023-09-02:
Status: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.3
Vbat: 2978 mV , Temp: 31.09°C, Humi: 57.79%, ID: 10, Reed switch counter: 0, flg: 0x04:r0/t0

Custom config HEX string: 55841000002804a9313104b400
(all default, only advertising changed to ATC1441)

server: ubuntu 22.04 with HA
BT: Intel® Dual Band Wireless-AC 8260
it only support BT4. 0, can it be there reason?

Maybe I should try to change Advertising interval and/or Coonect latency?

mmib, 2023-09-03:
Changed Advertising interval to 1500 and it fixed the issue. Now have concern about battery life with this settings.

pvvx, 2023-09-04:
BT4.0 has a small receiving radius.
The main hardware difference between BT4.0 and BT4.2 is an increase in receiver sensitivity and transmitter power.
The next addition is an increase in the size of the receiver-transmitter hardware buffer. But this does not affect the protocol used in the thermometer if BT5.0 is not enabled. 
Everything else is a software implementation. And so on until BT 5.4, except for the AoA/AoD hardware function.

mmib, 2023-09-04:
Thank you! Think it can be closed

closed 2023-09-04 by mmib