pvvx/ATC_MiThermometer #371: Cant connect to LYWSD03MMC w. Telink after setting Advertisement to 5000ms

g6094199, 2023-09-06:
the unit gests scanned

[image]

but the connection is always unsuccessfull. it stops reconnecting after 1st try.

[image]

Ubuntu 23.04
Chromium Version 116.0.5845.110
or
 Microsoft Edge Version 102.0.1245.7

pvvx, 2023-09-10:
These are problems in the Linux kernel.
Linux does not have full support for BLE standards. Bluetooth version 4.2, released in 2016, is not yet fully supported.
It is necessary to patch the kernel and use additional commands in the API.

pvvx, 2023-12-13:
Change the settings in Bluez to default values for the Bluetooth standard in /etc/bluetooth/main.conf.

closed 2023-12-23 by pvvx