pvvx/ATC_MiThermometer #372: After power cycle bthome broadcasts using random bindkey

agittins, 2023-09-10:
After I power-cycle my LYWSD03MMC's, they start broadcasting with what I think is a randomised bindkey, not the one I had configured it with. The bindkey appears to be stored in flash OK, and works if I re-configure the bindkey, but stops working after a power-cycle.

## Steps to reproduce:
- Configure LYWSD03MMC with:
  - advertising format: BTHome
  - adv flags: on
  - encrypted: on

- Send a bind key to the device
  - it reports it written and read OK (so looks like the flash write is good).

- disconnect
  - homeassitant then starts receiving encrypted advertisements, and homeassistant's bthome integration decrypts them OK.

- **Remove the battery** from the device for several seconds, then reinstall battery.
  - advertisements are still encrypted but homeassistant cannot decrypt them with the previously working bindkey.

- Connect to the device
  - "read bindkey" shows correct, original bindkey.
  - "write bindkey"
  -  disconnect
- Device sends encrypted advertisements, and homeassistant can decrypt them OK.

(note: currently Homeassistant's BTHome integration has a bug where it gets stuck trying to re-auth/reconfigure the device, I'll link to the PR to fix that once I've raised it)

## Guesses :-)

I notice that `init_ble` is called before `bindkey_init` but I don't know if that matters at all.

Since I can read the bindkey after a reboot it looks like the flash is working OK, but for some reason the device starts up not using the bindkey that is stored in flash.

pvvx, 2023-09-10:
You are using 2 bind-keys.
One from the official firmware and the second - recorded in memory by the flash_eep (custom firmware).
These are different keys.
Processing priority:
The first one taken is the key recorded during registration in MiHome.
If it is not there, then the key is taken from flash_eep.

agittins, 2023-09-10:
Ahh, thank you for your edit, that clarifies things for me a lot.

So I should use the "Show all mi keys" button then "!Erase all Mi Keys!" button to clear them?

I am curious why the custom firmware loads the mi key at all instead of just loading the flash_eep by default?

pvvx, 2023-09-10:
Because it can work with Xiaomi gateway if you know the keys and entered them during flashing, and also if after "authorization" the replace key with the previous one button is used.

closed 2023-09-20 by pvvx

agittins, 2023-10-06:
> In the new version 4.4 (current status: beta version), only the value from the "EEP BindKey" is taken as the binding key.

Fantastic, thanks! That feels a lot more logical. I've tried the beta on a couple of devices and it seems to be working well 👍🏼