pvvx/ATC_MiThermometer #374: [Feature Request] Receive temperature and display them

X-Ryl669, 2023-09-14:
Let's say you have 2 sensors node. Would it be possible that each node synchronize (somehow) and receive each other measurement to display them sequentially ?

For example:
| Time | Sensor 1 | Sensor 2 |
|-|-|-|
|0s | Display: 23°C | Display: 12°C |
|1s| Broadcast its temp | Broadcast its temp |
|2s|  Display: 12°E | Display: 23°E  (E used to mean external) |
|3s| Display: 23°C | Display 12°C | 

This way, one wouldn't need to have/run a gateway to see the exterior temperature.
I know that BLE advertisement are randomly emitted to avoid collision, but is there a way so have a pseudo random instead, so that the seed is known and the sensor can wake up and listen when the other sensor is advertising ?

pvvx, 2023-09-17:
This is possible only with the implementation of "[Bluetooth Basic Specification version 5.4](https://www.bluetooth.com/wp-content/uploads/2023/02/2301_5.4_Tech_Overview_FINAL.pdf)" ("Periodic Advertising with Responses (PAwR)" functionality).
But this standard is not supported by third-party systems. Linux and Windows does not yet fully support Bluetooth ver 4.2 either.

A "PAwR" or alternative implementation requires a more powerful power supply than the CR2032...

The tests (an alternative method compatible with bt4.2..5.0) showed a consumption of 20 µA in the mode after successful synchronization and 3..6 mA before synchronization. Synchronization due to collisions on the air may be disrupted, and a new synchronization requires dozens of ms with the receiver always on.

pvvx, 2023-09-18:
PAwR allows the server to transmit any readings to the thermometer for display.

"Old Test" is the implementation of the similarity of "PAwR", but with synchronization to another device.
Almost according to your description.

An external device transmits a typical advertisement.
The receiving device must accept this advertisement, but also transmit its own.
This requires synchronizing the transmission of ads from both devices. Otherwise, there will be reception issues during transmission.

According to the standard, the advertising period should contain an additional random delay. Not a **jitter**, since the new period begins with the end of the transmission of the past.

The available SDKs from Ralink (and many others) do not allow you to turn on the receiver for a given time without synchronizing with the transmission. But by redesigning, it was possible to make a suitable option - the transfer of advertising with switching to reception and completing reception through a timeout or receiving the necessary package of advertising.

The response reception window has a fairly long time interval (usually 15 ms) with a consumption of 6 mA. The probability of receiving a response is not equal to 100%, since collisions occur in radio communication.
And the longer the advertising period, the greater the additional delay -> a larger reception window is required.

As a result, if everything is already configured and synchronized, there is an increase in consumption by 7.5 ms * 6 mA.

Upon receiving the desired packet, the reception is interrupted and the advertising period is counted from it with a small deduction (-5..7 ms), since we do not know the real period of the transmitting device.
But if there was a loss (there was no reception in the window), then a new synchronization cycle is required. A missed reception will give an unknown time shift and will already require an increase in the reception window of 30 ms. Etc.
Further, if the advertisement is received, then the reception window is gradually reduced to 15 ms (or less, until the next disruption).

The initial, primary synchronization cycle requires an acceptance window of more than 1 advertising period. This is quite a large consumption.

closed 2023-10-23 by pvvx