pvvx/ATC_MiThermometer #376: Security?

luke-jr, 2023-09-16:
From reading the instructions, it sounds like there's zero interaction with the device itself to flash? What prevents a malicious actor from putting their own firmware on? Is it even possible to secure these devices?

luke-jr, 2023-09-22:
Ok, so basically just do the firmware switch in a safe place. Is it possible without using a website?

pvvx, 2023-09-22:
It is also possible to enter a PIN code in the "nRFConnect" program when connected to a thermometer by a special command.
You can download the catalog https://github.com/pvvx/pvv.github.io/tree/master/ATC_MiThermometer to the local disk and run "TelinkMiFlasher.html" without an internet connection.
Only these files are needed: `TelinkMiFlasher.html, styles.css, core.js`

Plus, it is possible to enter a PIN code using a variety of Bluetooth utilities.
Specification of the format of the [0x70 Set PinCode](https://github.com/pvvx/ATC_MiThermometer#primary-service-uuid-0x1f10-characteristic-uuid-0x1f1f) command - on demand or see the sources code.

closed 2023-09-29 by pvvx