pvvx/ATC_MiThermometer #378: Can't activate  LYWSD03MMC  HW:B1.5

DonTihi, 2023-09-19:
Hi!

When I'm trying to activate my LYWSD03MMC  it's stuck in  Activating now, please wait... 

This thermostat is already added to Mi Home, and refreshed it's firmware, and after that I deleted in Mi Home. But unfortunately its stucked again. I tried hard reset the thermostat az well. 

Could you suggest me a solution?

[image]

andreas-bulling, 2023-09-19:
Same problem here - I took them straight out of the box and tried to flash but it's not working. 

"NotFoundError: No Services found in device."

pvvx, 2023-09-19:
I don't have a file with the firmware version: 2.1.1_0159 and I can't check. I don't use MiHome.
If someone puts the file with the firmware version 2.1.1_0159 in open access, then adaptation is possible.

DonTihi, 2023-09-20:
Thanks pvvx! :) 

It's a very correct, clear, and quick answer! :) So there will be a solution, we just have to wait! :) Thank you!

vdende, 2023-10-05:
Ah, this is the same problem have and described with #388. I will close my ticket and wait for this one.

@pvvx Do you have the firmware 2.1.1_0159 in the meanwhile? If not, how can we extract it from our device?

Pancer1, 2023-10-09:
It's great tool, but I have the same issue with latest, updated Firmware. Stuck on "Activating now" and missing flash firmware buttons.

pvvx, 2023-10-10:
[image]

The "activation" and "advertising interval" algorithm has been changed to 2100 m s.

Flashing requires keys from "MiHome".
Set "Mi Token", "Mi Bind Key" and press "Login":
[image]

vdende, 2023-10-10:
Hi @pvvx , Thanks for looking into this. 

But I don't quite understand what to do. 
After I do 'Do Actiovation' the 'Device known id', 'Mi Token' and 'Mi Bind Key' are filled.

But When I press 'Login' it says 'Login Failed'.

[image]

thazro, 2023-10-10:
Same issue here. B1.4 on 2.1.1_0159. Stuck on "Activating now...". How can we get the MiHome keys?

Infnuk, 2023-10-10:
Same problem! Bindkey does not accept HA. Activating wait…

pvvx, 2023-10-11:
Firmware LYWSD03MMC version 2.1.1_0159 is temporarily not supported!
Xiaomi has implemented a new cipher or OTA firmware signature.

vasokot, 2023-10-29:
may be have any solution to downgrade firmware?

yaakovfeldman, 2023-10-30:
I recently bought this thermometer from aliexpress. Is there any way to check firmware version without trying to flash it? And if I try to flash and it turns out it does have 2.1.1_0159 and the flashing fails will I still be able to extract the bind key and use it with stock firmware? Thanks

VonalOrdu, 2023-11-02:
To go back to the old version (Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin );
To do this, you must remove your temperature sensor and connect it with the cables by following the steps below.
This is how I solved my problem.
I'm sorry for my bad english.

1. https://github.com/pvvx/ATC_MiThermometer/blob/master/Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin download
2. https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html 
3. [image]
4. [image]
5. File select Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin
6. Write to flash

pvvx, 2023-11-02:
[The USB-COM adapter writes the firmware in explorer. Web version.](https://github.com/pvvx/ATC_MiThermometer#the-usb-com-adapter-writes-the-firmware-in-explorer-web-version)
Connect only TX-SWS and GND wires.
[USBCOMFlashTx.html](https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html)

[image]

mesut3000, 2023-11-28:
Hi, any news about firmware 2.1.1_0159 ?

pvvx, 2023-11-28:
https://github.com/atc1441/ATC_MiThermometer/issues/298

gnr8shn, 2024-02-01:
Hi, I've bought what I thought was a LYWSD03MMC. It had the new firmware so I bought a USB <--> Serial adapter to be able to flash. But lo and behold the chip looks nothing like [[VonalOrdu's](https://github.com/VonalOrdu)](https://github.com/pvvx/ATC_MiThermometer/issues/378#issuecomment-1791370165) above. 

Does anyone know what this is and which soldering points correspond to VonalOrdu's?

Thanks all!
[image]

pvvx, 2024-02-01:
@gnr8shn - This device does not have BLE.

gnr8shn, 2024-02-01:
Ah what a shame thanks for the confirmation [pvvx](https://github.com/pvvx/ATC_MiThermometer/issues/378#issuecomment-1920591759).

ioukos, 2024-02-25:
Do we know when this new encryption/verification has been introduced ? I'm looking to buy LYWSD03MMC produced in 09/2023 and I'm afraid they won't allow me to upload your custom firmware.

Thanks

ioukos, 2024-02-26:
hi, @yaakovfeldman what is the date written at the back of the box ?
I've found some on a 2nd hand shop webside, date is 09/2023. I'm afraid it will embed new cypher.

gfan94, 2024-06-10:
> "advertising interval" has been changed to 2100 m s.

what was the advertising interval in the previous original firmware?

DittmannCz, 2024-07-12:
any solution to OTA flash to Zigbee?

braders2, 2024-10-08:
hello, any news in this topic?

kosielemmer, 2024-10-17:
I'd love to know too - any news?

robertolamm, 2024-10-21:
I am interested about this issue, also ... any news ?

ryck, 2024-11-12:
In case it's useful for anyone, this is how I was able to flash the latest ATC firmware using a USB2TTL dongle: https://github.com/atc1441/ATC_MiThermometer/issues/298#issuecomment-2465907616

RomanVolcic, 2024-11-14:
I did it with USB2TTL (just +, - and pin 14 to Tx). Works great. First one I downgraded to ver 1 then upload V 48, others I directly upload [ATC_v48.bin](https://github.com/pvvx/ATC_MiThermometer/blob/master/ATC_v48.bin)

[image]

lits101, 2024-11-16:
I'm also getting a blank screen after trying the above method. 

[image]

I'm not sure I understand the limitation of having to do this on a Windows PC? The flasher seems to work on macOS through Chrome.

juanjsebgarcia, 2024-11-26:
@lits101 I had the same issue flashing on macOS, native Windows machine worked perfectly first time.

falkenbt, 2024-12-05:
Did anyone succeed in flashing the firmware via an arduino uno? 
I followed this https://www.hackster.io/PatelDarshil/ways-to-use-arduino-as-usb-to-ttl-converter-475533 , shorting reset and ground. Then connecting TX to P14 and starting the flash of ATC_v48.bin via https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html - which reports success. 
But when checking the firmware version with https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html it still reports Version 2.1.1_0159. 
I also notice that on the arduino only the RX LED is blinking during the flashing process, not the TX LED. 
Power supply is working as I can see the device via bluetooth.
Any ideas what I could check?

Shino58, 2024-12-15:
I tried with arduino and is not working at all probably because of the led ? 
my sensor was blank after that so i ordered a proper adapter to flash the firmware

ulich, 2024-12-15:
I also had quite some trouble flashing any firmware, it always ended with a blank screen. I finally managed to make it work, see
https://github.com/atc1441/ATC_MiThermometer/issues/298#issuecomment-2543904906

Juanpermon, 2024-12-18:
Hi. I have the chance of buying some of these with zigbee flashed. Would it be possible to flash this firmware from that situation?

Thank you.

ICEAndreas, 2024-12-27:
Hi. I need help. The first flash didn't work. It stayed at 159. Then I changed the baud rate to 460800. After that the display was dark. Now, I have changed the setup as shown here: https://github.com/atc1441/ATC_MiThermometer/issues/298#issuecomment-1791414291 Now, the display is ok again, but I can no longer find it via Bluetooth. What can I do?

Oxi75, 2025-01-02:
Hi! I was semi-successful in reflashing LYWSD03MMC HW:B1.5 via UART. It works in general but my smart home central does not recognize it properly.
So now I'd like to go back to the original firmware but this seems to be a problem: Using the v1.0.0_0130 results in a working system for a few seconds. Then it seems to make a reset (all screen elements active).
The v1.0.0_0109 does also not work properly.

I assume that both FW versions are not compatible with the one that is required for my hardware. I have a 2nd device, bought at the same time than the 1st one. If I could read out the firmware I could try to upload it to the 1s device.
My question, because I was not able to find the answer by myself: How can I read out the firmware from the FLASH in way what it can be programmed by the this Webtool again:
https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html

Thank you!

WernerW1987, 2025-01-03:
> I did it with USB2TTL (just +, - and pin 14 to Tx). Works great. First one I downgraded to ver 1 then upload V 48, others I directly upload [ATC_v48.bin](https://github.com/pvvx/ATC_MiThermometer/blob/master/ATC_v48.bin)
> 
> ![Mijia 1](https://private-user-images.githubusercontent.com/122675299/386275339-dd08086f-e3a3-4251-8fba-6449c334ea8b.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU5MTMyNjYsIm5iZiI6MTczNTkxMjk2NiwicGF0aCI6Ii8xMjI2NzUyOTkvMzg2Mjc1MzM5LWRkMDgwODZmLWUzYTMtNDI1MS04ZmJhLTY0NDljMzM0ZWE4Yi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDEwM1QxNDAyNDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jMmU5NTdkNTc1NWNkYWJhMDZiMGMyYzllNDVmMDM3MGZhY2Y5MjdlMjQwZTdjZWU2ZTg3ZjU1ZGQxMDhiYmQ5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.BmG3Na299qQ3HAyuFtxvLHKpcGi4o7kBQnQO5MJHOJ0)

I Used this guide. work perfect.

wodixxxx, 2025-01-14:
I can also confirm that the 3 cable method worked perfectly for me today. I ordered a UART-TTL USB adapter with CH340G from Amazon. With this I was able to install the firmware “Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin” via cable.

after that i was then able to work again as usual with the Telelink Flasher. 

This worked 7 times today without any problems (HW version B1.5).

stillax, 2025-01-14:
Please have a look at this [https://github.com/atc1441/ATC_MiThermometer/issues/298](url).
There you will find the answer for flashing OTA.

journeyaway, 2025-03-30:
get token with https://github.com/Maxmudjon/Get_MiHome_devices_token
press Do activation button
“Device known ID”: empty
change “Mi token” to new correct token then  login

closed 2025-04-23 by pvvx