pvvx/ATC_MiThermometer #379: Thermometer detected by HomeAssistant but no values are visible

andreas-bulling, 2023-09-19:
Hi,

I successfully flashed to thermometers and they also got detected by HomeAssistant. However, the device only has one entity  "signal strength" but that one is empty. And no temperature/humidity values are visible.

pvvx, 2023-09-20:
Select the transmission protocol according to the integration used in the Home Assistant.
[TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html)

https://bthome.io/ -> ATC MiThermometer (V1)

https://github.com/custom-components/ble_monitor - support for all formats

...

andreas-bulling, 2023-09-20:
I am using the BT functionality directly integrated into HA - no custom component. From what I've read there are tons of people out there who have the same problem. Sensor doesn't get detected at all, or it does get detected by HA but then they are never asked to provide the bindkey.

I have left advertising type to "PVVX" (not encrypted). My understanding was that this should work when running the latest version of HA.

Two sensors now still (after waiting for 12 hours) only show the empty signal strength. A third one got detected but when I configured it it shows without any device or entity - not even the empty signal strength...

pvvx, 2023-09-20:
By default, only BTHome and Media(MiHome) work in Home Assistant (integrations: Bluetooth->BThome, Bluetooth-> Xiaimi_BLE )
[image] -> 
[image]

[image] -> 
[image]

"Passive BLE Monitor" works much better with good USB-BT adapters.
[image]

The integration of "Bluetooth" is much worse...

And write to HomeAssistant->Bluetooth, because everything works well for me, except for integrations related to "Bluetooth".

closed 2023-09-20 by pvvx