pvvx/ATC_MiThermometer #380: LYWSD03MMC as remote display from esphome - show ONLY remote data?

landrysik, 2023-09-19:
Hi, I have been experimenting with LYWSD03MMC as remote display for Home Assistant via esphome node.
https://esphome.io/components/display/pvvx_mithermometer.html

I tried chainging measure interval and min. lcd update interval but I can't find out how to display ONLY the data sent from esphome. The display keeps changing between the remote data and locally measured temperature and humidity. 
Is there any way how to show JUST the remote data and suppress the local measurements? I want to display outdoor temperature I get from another sensor and the switching between this and local measurements make the display confusing. 
Thanks a lot

pvvx, 2023-09-20:
The thermometer has the ability to work directly with a liquid crystal buffer. [The command "60xxxx..."](https://github.com/pvvx/ATC_MiThermometer#primary-service-uuid-0x1f10-characteristic-uuid-0x1f1f)
The transferred buffer is displayed until the connection is completed or the command "6100" is transmitted (switches to the standard display).
A description of the binding of the byte number and the bit number in the buffer to the LCD segments is given in the source headers for each type of thermometer. Example: https://github.com/pvvx/ATC_MiThermometer/blob/master/src/lcd_lywsd03mmc.c#L10

JS example: https://github.com/pvvx/pvvx.github.io/blob/master/ATC_MiThermometer/TelinkMiFlasher.html#L1988 :
```
const lcd_digcode = [0xf5,0x05,0xd3,0x97,0x27,0xb6,0xf6,0x15,0xf7,0xb7];
function lcd_clock() {
	let date = new Date();
	let hours = date.getHours();
	let minutes = date.getMinutes();
	let lcdb = new Uint8Array(7);
	lcdb[0] = 0x60;
	lcdb[1] = lcd_digcode[parseInt(minutes % 10)];
	lcdb[2] = lcd_digcode[parseInt(minutes / 10)];
	lcdb[4] = lcd_digcode[parseInt(hours % 10)];
	lcdb[5] = lcd_digcode[parseInt(hours / 10)];
	addClog("Send cmd (60): Set LCD data");
	settingsCharacteristics.writeValue(lcdb).then(_ => {addClog('Send data ok'); addLog("Send 'Send clock data on LCD' ok");});
}
```

pvvx, 2023-09-20:
In the new version 4.4 (current status: beta), the option to display only external data will be added when the display time value is set to 65535 seconds.  Permanent display, until reboot.

landrysik, 2023-09-20:
Excellent! Thank you. I've tested the beta firmware and it works like a charm!

closed 2023-09-20 by pvvx

udavst, 2023-09-30:
Minimum LCD refresh rate:  12.75 - это максимум, который устанавливается, как установить 65535?

landrysik, 2023-09-30:
@udavst You put those 65535 to the system you use to provide data for the display. NOT in the configuration of the firmware itself. In the case of esphome you put   `validity_period: 65535s `