pvvx/ATC_MiThermometer #386: Temperature and Humidity offset - no decimals

remezen, 2023-09-28:
Hello.
I want to make an offset, e.g. 0.4 degrees.
When I put this number in a field in TelinkMiFlasher.html and press save, then load config - the offset is "0".
Can't understand - is it rounded in TelinkMiFlasher.html, but saved, or not.

pvvx, 2023-09-29:
After entering the offset value, select "Send Config".
[image]

The entered offset value is always shown.
After saving the configuration, it is automatically re-read and displayed.

To force the configuration to be read, "Get config" is used.
[image]

pvvx, 2023-09-29:
The step of the offset values is 0.1 unit

closed 2023-10-23 by pvvx

maiandrei, 2024-01-03:
Is there any chance to have, somewhere in the future, offset steps of 0.01°C ?

Thanks for any answer, even if not positive.
You unleashed the true power of these sensors. Thanks for that.

maiandrei, 2024-01-07:
I understand now that field is 8bit (-127 to +127), meaning that offset can have values from -12.7°C  to 12.7°C,
Is it too hard to switch that to -1.27°C to +1.27°C? ALL my temperature offsets are in the +/- 0.2°C now.

pvvx, 2024-01-07:
Some users ask for values greater than +-5°C. :)

maiandrei, 2024-01-07:
Thank you so much, you gave me hope. 
I've put together, in box and bubblewrap, more sensors to 'calibrate them'. The deviation from average was always less than 0.2°C.
That average was also in the +/- 0.1°C deviation from a UFH thermostat sensor and a zigbbe TH sensor. That tells me these Xiaomi sensors are really accurate.
I don't know why people want to shift their reported values with +/-5°C, maybe for specific applications :)
I also am aware that such level of detail might be overkill for most applications, but I want to use these sensors for UnderFloorHeating and I need to compare return temperatures from different circuits of the same room and... i know, i'm crazy :))

I will look into source codes, never did that before, will try to understand the whole process and build a version.
Thanks for your patience!

PS. MJWSD05MMC is a bit less "jumpy" than LYWSD03MMC - most probably thanks to the battery (CR2450 vs CR2032) - about max 0.04°C vs 0.08°C, as I can see in my readings.

pvvx, 2024-01-07:
Accuracy is increased by the presence of a capacitor in the power supply circuit.
But battery voltage also affects the sensor readings. And battery voltage depends on temperature. And it depends on the battery charge level and the pulse load of the SoC operation.

maiandrei, 2024-01-07:
Is it worthy to increase the capacitor? Maybe I will give it a try.
In this picture we can see MJWSD05MMC (purple) vs LYWSD03MMC (blue).
[image]

pvvx, 2024-01-07:
Also, the SHTv3 sensor responds faster than the SHT4x. SHT4x seems to already use averaging.

pvvx, 2024-01-07:
The “thermostat” in the HA cannot work with 0.01C.
[image]

pvvx, 2024-01-08:
> I don't know why people want to shift their reported values with +/-5°C, maybe for specific applications :)

https://github.com/devbis/z03mmc/issues/85#issue-2022805111