pvvx/ATC_MiThermometer #387: Uncaught ReferenceError

ghigo, 2023-09-30:
The webpage console shows the following error

TelinkMiFlasher.html:1139 Uncaught ReferenceError: hwver_id is not defined
    at menuUpgrade (TelinkMiFlasher.html:1139:44)

Unable to flash a build

closed 2023-12-23 by pvvx

Danstiv, 2025-05-25:
Same problem on Telink Flasher v10.1
Configuration fields and firmwares are not showing.
What can I do with this?
```
01:07:37: Searching for devices
01:08:00: Connecting to: ATC_6C557D
01:08:06: Hardware Revision String: B1.5
01:08:06: Software Revision String: V4.7
01:08:06: Firmware Revision String: github.com/pvvx
01:08:06: Detected custom Firmware
```
@pvvx

pvvx, 2025-05-25:
Found:
```
TelinkMiFlasher.html:847 Uncaught ReferenceError: hwver_id is not defined
```
Fixed in v10.2.