pvvx/ATC_MiThermometer #390: Registration failed! on 3 out of 4 LYWSD03MMC

elfnor, 2023-10-10:
I bought a batch of 4 LYWSD03MMC with CMIIT ID 2019DP8115 
I successfully flashed the first one from 
https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html
using Edge browser on win10 after trying firefox and opera on both Linux and windows.

The next 3 connect OK, and the flasher reads the battery, temperature and humidity but on clicking the "Do Activation" button I get "Registration Failed!"
```
5:56:14 pm: Searching for devices
5:56:30 pm: Connecting to: LYWSD03MMC
5:56:36 pm: Hardware Revision String: B1.5
5:56:36 pm: Software Revision String: 0130
5:56:36 pm: Firmware Revision String: 1.0.0_0130
5:56:36 pm: Detected Mi device
5:56:36 pm: Connected
5:57:16 pm: Activating now, please wait...
5:57:20 pm: Registration failed!
```
The flashed device also reports Hardware Revision String B1.5

I checked the reported battery voltage for each and there doesn't seem to be a pattern.
A 3029 mV registered and flashed
B 2566 mV won't register
C 3204 mV won't register
D 3155 mV won't register

Clues or a step I missed? I can't see anything I'm doing differently than the first time.

slydeur, 2023-10-10:
I experienced the same. First one (a few days ago) was no problem.

guidokieboom, 2023-10-11:
I experienced exactly the same. Tried with Edge and Chrome on Window11. Only difference is 
Hardware Revision String: B2.0 in my case. Retry after battery remove did not change the result.

closed 2023-10-11 by pvvx