pvvx/ATC_MiThermometer #391: LYWSD03MMC weird battery level behavior

dmig, 2023-10-11:
I have LYWSD03MMC revision B1.4 running firmware V4.3.

At some point humidity level got replaced by battery level, showing 15% or lower.
Humidity level will get back to the screen only on bluetooth connection attempt, also this resets low battery status (I believe it shouldn't).

Flasher often fails to connect or read configuration -- I suppose battery is too weak.

2 things here I see wrong:
- battery level replaces humidity (original firmware only shows indicator)
- low battery gets reset on BT interaction

pvvx, 2023-10-11:
https://github.com/pvvx/ATC_MiThermometer/blob/master/src/lcd.c#L89C22-L89C36

dmig, 2023-10-19:
> For the stability of the connection in LYWSD03MMC, it is necessary to install capacitors in the power supply circuit that were not installed on the printed circuit board due to the stinginess of the manufacturer.
Thanks! Will take a look.

closed 2023-10-23 by pvvx