pvvx/ATC_MiThermometer #393: Update error: NotSupportedError: GATT operation failed for unknown reason.

floringhimie, 2023-10-19:
Hi and thank you for your work.
I have 10 LYWSD03MMC devices, I already flash 9 of them with this method but last one fails.
I tied to change the battery, reset GND+RST, sending the 0x56 command but the same behavior. 
Any idea?
Thank you!

10:58:44 AM: Searching for devices
10:58:50 AM: NotFoundError: User cancelled the requestDevice() chooser.
10:58:50 AM: Reconnect 1 from 5
10:58:55 AM: Searching for devices
10:58:57 AM: NotFoundError: User cancelled the requestDevice() chooser.
10:58:57 AM: Reconnect 1 from 5
10:58:58 AM: Searching for devices
10:59:02 AM: Get Advertising MAC (Web Experimental Features!)
10:59:02 AM: Connecting to: LYWSD03MMC
10:59:10 AM: MiVer5, DevID: 0x055B-LYWSD03MMC, FnCnt: 1, CtrID: 0x5830 Not bound, Standard certification, MAC: A4C138A0006A, Capability: 0x28 - front binding, IO: 0x0001
10:59:15 AM: Hardware Revision String: B1.4
10:59:15 AM: Software Revision String: 0109
10:59:15 AM: Firmware Revision String: 1.0.0_0109
10:59:15 AM: Detected Mi device
10:59:15 AM: Connected
10:59:18 AM: Activating now, please wait...
10:59:22 AM: Registration successfull
10:59:22 AM: Send Login, please wait...
10:59:23 AM: Received device infos are correct
10:59:23 AM: Login successfull
10:59:25 AM: Load firmware file 'ATC_v43.bin'...
10:59:25 AM: File: ATC_v43.bin
10:59:25 AM: File size: 81540 bytes
10:59:25 AM: Count: 5097
10:59:27 AM: Start DFU
10:59:33 AM: Update error: NotSupportedError: GATT operation failed for unknown reason.
10:59:33 AM: Disconnected.

pvvx, 2023-10-20:
Maybe this will help.
Restart the device from which the update is being performed. Or reset the Bluetooth adapter...

landrysik, 2023-10-23:
I have the same issue. From what I discovered I have problem only with hardware version B1.6 (works on 1.7, 2.0 and 1.4). Unfortunatelly I don'T have different firmware versions on those 1.6 thermometers - all have 4.2
I can't update any of them (at least 3 specimens) Tried different android phones + windows laptop. Ends with the same error
Was thinking about bad power but I tried to even power them from 3.3V lab supply but still fails.

The update fails in the moment it gets stuck on "sending block nr 7". When I update working one (HW2.0...), it gets stuck on "sending block nr 7" for a second and then continues. HW1.6 does not continue and fails

landrysik, 2023-10-23:
Having similar issue here, just the error message is "NotSupportedError: GATT Error Unknown". From what I discovered I have problem only with hardware version B1.6 (works on 1.7, 2.0 and 1.4). Unfortunatelly I don'T have different firmware versions on those 1.6 thermometers - all have 4.2
I can't update any of them (at least 3 specimens) Tried different android phones + windows laptop. Ends with the same error
Was thinking about bad power but I tried to even power them from 3.3V lab supply but still fails.

The update fails in the moment it gets stuck on "sending block nr 7". When I update working one (HW2.0...), it gets stuck on "sending block nr 7" for a second and then continues. HW1.6 does not continue and fails

pvvx, 2023-10-23:
OTA (DFU) works the same in all versions.
Maybe something happened to the flash memory (need to erase all flash memory and remove the write protection attribute).

Ircama, 2023-10-26:
I had the same issue after flashing the firmware of a new device. Also, the device got systematically stuck on each connection attempt and needed a reset by removing the battery. Both the browser and this command failed: `python3 -m atc_mi_interface.atc_mi_config -m A4:C1:38:AA:BB:CC -vicg`.

By simply changing the PC (or using e.g. an Android device), the configuration works. It appears that the PC that performed the flash might happen to remember some old attributes sometimes. I have not had time to understand why and how to fix/clean that on the other PC.

```
13 Model Number String LYWSD03MMC
19 Hardware Revision String B1.7
21 Software Revision String V4.5
```

pvvx, 2023-10-26:
>  It appears that the PC that performed the flash might happen to remember some old attributes sometimes.

According to the standard, the adapter is required to remember the entire attribute table by binding it to the MAC

pcmihnea, 2023-11-02:
I'm having the same issue with a LYWSD03MMC B1.6, Software Version: 4.2. Trying to update to 4.4 or 4.5 results in the same behaviour, where it disconnect at block 7. Tried with multiple devices, each with different OSes. Flashed six LYWSD03MMC B1.7 to 4.5 with no issues. All batteries are new, and the devices next to the host PC/phone. Does anyone have any ideas for a fix?

pvvx, 2023-11-02:
Have you tried to reduce the "connection latency" to 1 second or lower, disable BT5.0 and reconnect?

pcmihnea, 2023-11-02:
I issued a default config, [133, 16, 0, 0, 40, 4, 169, 49, 49, 4, 180], so connection latency is set to 1000ms, while BT5+ PHY and LE Long Range are both disabled. Still no luck...

fakemanhk, 2023-11-13:
I am getting the same problem, looking at above comments, where should I set connection latency? (I am using Android phone to flash)

```
4:16:26 PM: Searching for devices
4:16:30 PM: Get Advertising MAC (Web Experimental Features!)
4:16:30 PM: Connecting to: ATC_C2FD83
4:16:31 PM: MAC: A4C138C2FD83, Bat: 100%, Vbat: 2966 mV , Temp: 25.34°C, Humi: 26.86%, Count: 3, Flg: 4
4:16:45 PM: Hardware Revision String: B1.5
4:16:45 PM: Software Revision String: V4.4
4:16:45 PM: Firmware Revision String: [github.com/pvvx](https://github.com/pvvx/ATC_MiThermometer/issues/github.com/pvvx)
4:16:45 PM: Detected custom Firmware
4:16:47 PM: Hardware Version: LYWSD03MMC B1.5, Software Version: 4.4, Sensor: SHTC3 (SHTV3)
4:16:47 PM: Custom config HEX string: 55051000002804a931318ab4
4:17:20 PM: File: z03mmc.bin
4:17:20 PM: File size: 130004 bytes
4:17:20 PM: Count: 8126
4:17:37 PM: Start DFU
4:17:42 PM: Update error: NotSupportedError: GATT Error Unknown.
4:17:42 PM: Disconnected.
```

pcmihnea, 2023-11-14:
> I am getting the same problem, looking at above comments, where should I set connection latency? (I am using Android phone to flash)
>
> ```
> 4:16:26 PM: Searching for devices
> 4:16:30 PM: Get [...]

Hi, if you're using the Telink flasher (https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html), scroll down to the configuration fields once connected to the sensor, and edit the 'connect latency':
[image]

fakemanhk, 2023-11-22:
Gotcha, now I have it flashed (but after flashing my phone/laptop bluetooth can't see it any more, I guess this is correct, right?)

olivluca, 2023-12-22:
> I issued a default config, [133, 16, 0, 0, 40, 4, 169, 49, 49, 4, 180], so connection latency is set to 1000ms, while BT5+ PHY and LE Long Range are both disabled. Still no luck...
> 
> UPDATE: got the OTA working once reduced connection latency to 520ms (oddly 500ms didn't work) and left all other parameters to default config values. Thanks for the help!

Thank you, having the same issue trying to upgrade my sensors running firmware 1.2, this fixed it for me.

closed 2023-12-23 by pvvx

BenJamesAndo, 2024-02-18:
I had the same error. I was on version 4.4 trying to upgrade to version 4.6. None of the above solutions worked.
But I did get it to work by flashing the Original Xiaomi firmware and then flashing to 4.6.

scottcopus, 2024-09-13:
In case this helps anybody, I had the BT pin code set and kept getting this error too. But after I cleared the pin code (000000), the firmware upgrade from 4.3 to 4.7 worked fine.
_Hardware Version: LYWSD03MMC B1.5, Software Version: 4.3, Sensor: SHTC3 (SHTV3)_