pvvx/ATC_MiThermometer #394: Home Assistant and SSD TBW.

pvvx, 2023-10-24:
recorder in configuration.yaml:

```
recorder:
  purge_keep_days: 31
  commit_interval: 60
```

[image]

```
cat /sys/block/mmcblk1/stat | awk '{printf "%.3f\n", $7*512/1048576}'
cat /sys/block/nvme0n1/stat | awk '{printf "%.3f\n", $7*512/1048576}'
smartctl -A -j /dev/nvme0 | awk '/"data_units_written"/ {printf "%0.3f\n", $2*512/1000000}'
```

View all active disk writes in the system:
```
fatrace -t -f W
```

tigers75, 2023-11-10:
Sorry if the question seems dumb: this only affects the SSD if the BT receiver is connected directly to the machine running HA, am I correct?
Also in the Warnings I see reported that:

_Warnings

Requires BT adapter with USB2.0 HS. Otherwise, there will be many gaps in the reception. Do not use ESP32xxx - it cannot work with BLE at normal speed and drains the batteries of all BLE devices when connected! ESPHome does not work with Bluetooth 5.0 and misses a lot of advertising packets from BLE devices._

But I see many in the forums using ESPHome to connect (that's my config: an ESPHome mainly dedicated to BT devices (3 of them actually) + BTHome protocolo + Home Assistant). Is this safe or this will drain the BLE battery (still don't understand how since the ESP should be in "passive" mode and only get data when the device sends it out)?

pvvx, 2023-11-10:
> ESP should be in "passive" mode 

_misses a lot of advertising packets from BLE devices_

> Sorry if the question seems dumb: this only affects the SSD if the BT receiver is connected directly to the machine running HA, am I correct?

When using Bluez.  This is not the main problem.
The main problem is unoptimized database recording in HA and affects when you have many devices.

tigers75, 2023-11-10:
Thanks, so I can just exclude the sensor from the recorder: in config.yaml and I'm going to be fine?

pvvx, 2023-11-10:
Then there will be no measurement history.

For recording, all unnecessary and uninteresting entities, such as reception levels and others, are disabled. Store depth limitation for one month and data saved at one minute intervals.

thegillimonster, 2023-11-15:
Pardon my confusion: is this to say that there is no good way to connect BLE sensors to Home Assistant? Based on what I can see here, using a BT dongle on the HA machine ruins the SSD, and using an ESPHome device in the middle misses advertisements and wastes battery on the sensors. I have ten LYWSD03MMC, is there any solution that truly works?

tigers75, 2023-11-15:
If I understood this correctly you can exclude the BLE sensor form recording to the DB, and if you need an history record for any of them create a dummy sensor that updates every once in a while, like when the values changes for more than 0.5 °C, or more than 1% HR.
That's my config.yaml setup, the "entity_globs:" section accepts wildcards so it should be pretty easy to filter the BLE sensors:
`recorder:
  purge_keep_days: 14
  # limita la scrittura ogni 60 secondi per stressare meno l'SSD
  commit_interval: 60
  exclude:
    domains:
      # - sensor
      - automation
      - script
      - light
      - camera
      - delay
      - group
      - media_player
      - timer
    entities:
      - light.ingresso
    entity_globs:
[...]
      - sensor.termometro*`

pvvx, 2023-11-15:
This still does not allow Home Assistant to work for a long time on eMMC and especially on SD cards. Only on SSD or HDD.
With SSDs, it is additionally necessary to provide power redundancy, due to the lack of a “fast parking” function for SSDs in Linux. SSDs themselves have this function, but there is no implementation in Linux.

tigers75, 2023-11-15:
> This still does not allow Home Assistant to work for a long time on eMMC and especially on SD cards. Only on SSD or HDD. With SSDs, it is additionally necessary to provide power redundancy, due to [...]

That' s a generic HA anyways, not related to BLE devices in particular (I have switched to SSD a long time ago also gaining some speed at boot). My plan is also to have a full HW backup ready to switch in in case of failure (another raspberry, with another SSD); at the cost SSD are now I can consider it like some sort of "consumable", to switch completely every 2/3 years.
I'm interested in the power redundancy thing, which I was not aware of: what am I risking right now using a SSD without it? I don't think it's physical "parking" since it doesn't have moving parts, so what's the problem?

pvvx, 2023-11-15:
> I don't think it's physical "parking" since it doesn't have moving parts, so what's the problem?

Failure due to loss of markings or programs due to a power failure. Server SSDs have additional capacities and other elements and a program for emergency reset of vital data. This is not the case in household ones.

https://www.phoronix.com/news/NVMe-Abrupt-Shutdown-Linux

smartctl -a /dev/nvme0
```
Unsafe Shutdowns:                   3
```

ffries, 2023-12-15:
Hello,

You don't have to worry about that. First, under GNU/Linux, the data is not committed to disk on every write. Secondly, the SSD firmware decides where to write data and data is not always contiguous. The firmware choose to write where ZEROs are written. Thirly, you can trim your SSD using Linux trim or on file removal, which gives SSD firmware more space to choose.

pvvx, 2023-12-15:
@ffries 
Only HA is installed on the SSD.
[image]

eMMC (system) and NVMe (Home Assistant)
```
UUID=xxxxxx-xxxx-xxxx-xxxx-xxxxxxxx	/		ext4	defaults,noatime,commit=600,errors=remount-ro,x-gvfs-hide	0	1
UUID=xxxxxx-xxxx-xxxx-xxxx-xxxxxxxx	/mnt/nvme	ext4	defaults,noatime,commit=600,discard	0	2
```

> Last if you are really worried you may use a RAM disc for a folder or some files.

https://stackoverflow.com/questions/65957677/bluez-change-local-storage-directory
> Symlinking does not work for me. The service fails with the error "Not a directory".

Ircama, 2023-12-17:
> Replace the SD card on RPI 5 with a new one every month or after a couple of weeks.

With Raspberry Pi, enabling the Overlay Filesystem (which uses the RAM to store changes) allows setting the SD card in read-only mode (or defining an overlayed read-only partition for the root filesystem and a non-overlayed RW partition for the data). It can be configured via raspi-config. With the Overlay Filesystem, the system can be turned off without shutdown (but if a RW data partition is defined, it will not be protected).

pvvx, 2023-12-17:
Home Assistant is very slow on an SD card. It's the same on an SSD drive via ATA.
It takes a very long time for the graphs to be drawn and various errors constantly appear.
Only on NVMe and eMMC Home Assistant starts to work decently. Requires disk read-write speed of 1 GB/sec.

ffries, 2023-12-28:
I am running HA on an Nvme and SSD on different systems. The NVMe is not comparable to an SD card. The NVMe has an internal controller that prevents the card from wearing to quicly. And data is not written instantly. All this discussion is pointless and you should close this issue.

pvvx, 2023-12-28:
@ffries - what are you talking about?
`commit=600`

convexshiba, 2023-12-29:
The BLE devices do update quite often.

I assume it will ameliorate (eliminate?) the excessive write situation if I convert it to be using Zigbee instead of BLE?
(ref: https://smarthomescene.com/guides/convert-xiaomi-lywsd03mmc-from-bluetooth-to-zigbee/)

It states that in Zigbee firmware the sensor only reports every 5 minutes, which I hope can be made configurable.

sebastianciupinski, 2024-01-16:
I wish to understand what is this case about...

AlexKratky, 2024-02-08:
I don't think I fully understand this issue. I have raspberry pi 4 with 32 GB sd card and 4 thermometers in BTHome v2 mode. Does this affect me? Should I worry about lifespan of the sd card? I was expecting to changed it every year to be safe, but reading this meaning I should do it more frequently or switch to HDD? I do not need so detailed history (only maybe for week, then its okay to have average value for lets say 1 hour period, is there an easy way to setup it like that?)

pvvx, 2024-02-08:
> The BLE devices do update quite often.

Zigbee Tyua:

[image]

pvvx, 2024-02-08:
> I do not need so detailed history (only maybe for week, then its okay to have average value for lets say 1 hour period, is there an easy way to setup it like that?)

The entire database is rewritten every day to remove old data.
When using an SD card, there will be problems with a couple of dozen of any sensors. Errors will occur due to very low performance.
For comfortable use of HA, a media speed of 1 GB per second is required.

Deoptim, 2024-03-01:
On Debian(Armbian tested) you can bypass the option to write the BlueZ cache to disk in two ways:
**First one:** recompile BlueZ with the modified STORAGEDIR option in a folder say **_/tmp/lib/bluetooth_**

**The Second** way is more simple:
Already on a working system (Armbian tested):
1. Install **bindfs**:
`sudo apt install bindfs`
3. Make an entry in **/etc/fstab** with content like:
`bindfs#/tmp  /var/lib/bluetooth  fuse  user,comment=x-gvfs-hide  0  0`
_(pay attention to the option **comment=x-gvfs-hide** - it hides from BlueZ that this is a binded folder)._
5. Stop the bluetooth service:
`sudo systemctl stop bluetooth.service`
6. Delete the old folder before rebooting (this is important):
`sudo rm -Rf /var/lib/bluetooth`
7. Reboot

Deoptim, 2024-03-01:
My Orange Pi (Armbian **only SD-card**) + Home assistant were disabled logging to the memory card, the memory card by the way is used Smasung High Endurance. 
In Home assistant instead of standard database MariaDB is used (it is strongly advised to install SD-card less host writes to it, it can be installed via Add-ons store).
In the configuration for this database is used database storage no more than a month and disabled some records of sensors such as System Monitor - the database itself is compressed in memory and takes up little space - about 90MB for a month (it is three sensors LYWSD03MMC, BTH01, TH05 and so on).
[image]

```
recorder:
  db_url: !secret mariadb_url
  purge_keep_days: 60
  commit_interval: 1200
  exclude:
    entities:
      - sensor.last_boot
      - sensor.processor_temperature
      - sensor.processor_use
      - sensor.memory_use_percent
      - sensor.disk_use_percent
      - sensor.date_time
      - sensor.home_assistant_supervisor_cpu_percent
      - sensor.home_assistant_supervisor_memory_percent
```

pvvx, 2024-03-01:
> about 90MB for a month (it is three sensors

90MB per month / three sensors = 30MB per sensor?
I have 89 or more devices in one of the HAs.
This will come out to 89*30 = 2670 MB.
The actual size is 6059 MB. The database is over a year old.

pvvx, 2024-03-06:
> On Debian(Armbian tested) you can bypass the option to write the BlueZ cache to disk in two ways: **First one:** recompile BlueZ with the modified STORAGEDIR option in a folder say **_/tmp/lib/bluetooth_**

@Deoptim According to the "Uptime MMC writen" graph, the recording decreased by 320 megabytes per day.

Deoptim, 2024-03-06:
@pvvx
If you want to change the database to MariaDB -  [here I tried this way of conversion SQLite to MySQL](https://4pda-to.translate.goog/forum/index.php?showtopic=871505&st=42580&_x_tr_sl=ru&_x_tr_tl=en&_x_tr_pto=wapp#entry128823246) ([original source](https://4pda.to/stat/go?u=https%3A%2F%2Fgist.github.com%2FPattaFeuFeu%2Fc4475457854f42f64f21268777d64d87%23file-ha_migrate-sqlite-to-mysql-md&e=128823246)).
There is even more economical option (for SD-Card or eMMC host record) to use [mariadb-inmemory Add-on](https://github.com/lmagyar/homeassistant-addon-mariadb-inmemory) ([discussion](https://community.home-assistant.io/t/in-memory-mariadb-mysql-add-on-for-recorder-history-integration/281791)).
In a month I will post the results of the comparison with the original database "home-assistant_v2.db" / local MariaDB / In-memory MariaDB.

50494554524F, 2024-03-21:
i'm using this trick, i'm running homeassistant os + sd + hdd on a rpi... go to /var/lib/bluetooth/MACADDRESS and just run

rm -r MACADDRESS && > cache

no more writes to the microsd and all (seems) working at the moment, no needs to disable active scan (i have a switchbot bot to activate and requires the active scan to work)

(i just started studing linux so i did all by my self, very happy :D)

pvvx, 2024-03-21:
These are all problems with Bluez and it is not clear why it is used in Home Assistant. Bluez does not yet fully support Bluetooth 4.2 (2014). Modern Bluetooth has version 5.4.

Deoptim, 2024-04-24:
@pvvx
Here's a late one, but here's a little statistic nonetheless:
[Russian Language](https://4pda.to/forum/index.php?showtopic=871505&view=findpost&p=129852450)
([Google Translated](https://4pda-to.translate.goog/forum/index.php?showtopic=871505&st=44960&_x_tr_sl=ru&_x_tr_tl=en&_x_tr_pto=wapp#entry129852450))

pvvx, 2024-04-24:
@Deoptim 
Для тестов HA 20 октября 2023 был установлен новый NVMe 256G (PCIe MAXIO MAP1202A-F1C  2 x YMC3G001Tb62BA1C0)  в новую NanoPi-R5S (RAM: 4GB LPDDR4X, 16GB eMMC). Сама база HA переехала с прошлой платформы, с SSD.
На сегодня:

[image]

Со старта HA работает постоянно (питание автономное). Кроме HA ничего другого не установлено, docker не используется. Система на eMMC, HA c базой на NVMe. Производилось несколько обновлений всего ПО, но это копейки по сравнению с суточной записью HA.
Датчиков и прочих устройств около сотни - 3 USB адаптера: 2 BLE, 1 Zigbee. В настройках HA приняты меры фильтрации записи лишнего в базу...

Если отработает год - полтора, то всё равно менять на новую платформу, т.к. для комфортной работы требования к производительности и ресурсам всегда растут.

pvvx, 2025-03-04:
It's still working.

[image]

The price is low, can always replace the NVMe with a new one...

trablin45, 2025-06-10:
is this still an ongoing problem as today? its better not use bluetooth on HA ?