pvvx/ATC_MiThermometer #395: Issue with the accuracey of the CR2032 graph and what's actually happening.

420pootang69, 2023-10-24:
I'll keep this brief. 

I initially was setting my units to advertise (using ATC profile) every 10 seconds, but after seeing the graph showing that there is no/little point in this I switched to 2.5 seconds.

Two months later and the unit at 2.5 seconds is showing a battery of 80%, but the ones at 10 seconds are mostly still showing 100% battery.

Am I reading that chart incorrectly, or is the chart wrong?

pvvx, 2023-10-24:
This is the calculation for Xiaomi LYWSD03MMC B1.4 with a 220 mAh CR2032 battery (+25С).
Taking into account the installed capacitor on the printed circuit board in the power circuit.
Without capacitors, the battery operates up to 60%.
The diagram is based on real measurements of sleep currents and circuit activity at a transmission level of +0dB.
In the settings, only the interval is changed. The rest of the settings are default.
The external BT-BLE logger works in passive scanning mode (!).

420pootang69, 2023-10-25:
Thanks for your detailed response.
Honestly I was going with the idea that they're just terrible CR232s they come with as stock and when I change them, I might get a better lifespan. They are ridiculously cheap modules after all.
Yes, I am using them in advertising mode and passively receiving the messages, not connecting to them.

> (Failure due to high humidity when working on the second battery. Sent to the trash heap.)

Can you expand on this one? I put one outside during wet Irish weather where the humidity would have been at 90%. It failed and seems to be in a boot loop or something. Just wondering if I'm wasting my time putting another outside and should use something else for outside measurements.

pvvx, 2023-10-25:
Our prevailing humidity is >85%.
In summer there are two thermometers in the greenhouse. Every night they are covered with dew, in the form of dripping drops... :)

pvvx, 2023-11-19:
Xiaomi LYWSD03MMC V1.9 test on the cheapest CR2016 battery
[image]

sezlony, 2023-12-07:
how is it possible, that battery still works for days (or even for weeks) after reaching 0% ? this annoyes me more than the (relative) quick discharging. could it be a FW bug or is it just me?

[image]

pvvx, 2023-12-07:
CGPR1 Qingping Motion & Light: 

[image]

[image]

sezlony, 2023-12-07:
are you such offensive to anyone who highlights a personal experience or just having a bad day?

whatever is the case relax and lay back, at the end of the day you are no more than a human, like anyone else, don't take yourself too serious! 😆

sezlony, 2023-12-07:
> There is a personal blog for this.

you're really pathetic with this hostility... go take a brake! [while I stand with Ukraine!](https://github.com/vshymanskyy/StandWithUkraine)

sezlony, 2023-12-09:
Congratulations, you're just as fucking retarded like your Putin! 🤣🤣🤣🤣

keeping the propaganda while removing other's opinions/experiences is sooo much retarded russian... 😆

### Delete this too, so people will know you're truly a suppressive fucker!

[image]

[image]

sezlony, 2023-12-09:
delete this fucking russist

[image]