pvvx/ATC_MiThermometer #397: [Feature] Broadcase firmware version

geever, 2023-10-27:
I would like to keep my LYWSD03MMC-s up to date, it would be nice if the firmware version was broadcasted, so I could take that, and the latest release version from your GitHub repo to make a binary_sensor in my Home Assistant showing if I need to update.

i'm using the BTHome encrypted beacon, but other beacons could also benefit from such feature. I see no version related data type in the BTHome specification, but it should surely be possible to use a text data type for this.

pvvx, 2023-10-28:
Not welcome, as such an addition will increase battery consumption for everyone.
The SW version is in the standard attributes of the BLE device.

tmugan, 2023-11-14:
Duplicating here in case anyone would like to upvote
https://community.home-assistant.io/t/make-ble-softwarerevisionstring-available/640675?u=tmugan

ilgrank, 2023-11-27:
As @pvvx says, the life of the BLE devices is directly dependant on air time.
broadcasting more data than strictly necessary has a direct inpact on battery life.
Simply saving 2-3 bytes means *months* of additional battery life out of a CR2032 battery.

closed 2023-12-23 by pvvx