pvvx/ATC_MiThermometer #398: Reading fluctuations

melwinek, 2023-10-29:
Thermometers with hardware version B1.4 have reading fluctuations.  This is not the case in B1.7.  Is this typical for this version?  Can something be done about it?

Normal:
[image]

Fluctuations:
[image]

melwinek, 2023-10-29:
Can be closed.  Disabling "Sensor in Low Power mode" solves the problem

pvvx, 2023-10-29:
B1.4 uses a different sensor (SHTC3).
Compared to the new versions (SHT4x), this sensor has a higher consumption and a faster response to changes in the measured parameters of the medium.
In low power mode, SHTC3 gives inaccurate readings. And for SHT4x, the polling algorithm changes, which does not affect accuracy.

closed 2023-10-29 by pvvx