pvvx/ATC_MiThermometer #400: MJWSD05MMC ESPHome

V-VoRteX-V, 2023-10-29:
How to display temperature from HA sensor? With LYWSD03MMC everything is fine using commands it.print_bignum or  it.print_smallnum https://esphome.io/components/display/pvvx_mithermometer.html 
But MJWSD05MMC works strangely, it displays incorrectly

pvvx, 2023-10-30:
MJWSD05MMC uses a different format and displays only 1 number.
 number: -999.50 .19999.50, in 0.01: -99950..1999950
```
 *  -999	(-9.99)	[ -999][ -9.99]
 *   -99	(-0.99)	[ -99 ][ -0.99]
 *    -9	(-0.09)	[  -9 ][ -0.09]
 *     9	(0.09)	[   9 ][  0.09]
 *    99	(0.09)	[  99 ][  0.99]
 *   999	(9.99)	[ 999 ][  9.99]
 *  9999	(99.99)	[ 9999][ 99.99]
 * 19999	(199.99)[19999][199.99] 
```

https://esphome.io/components/display/pvvx_mithermometer.html - this only works for Xiaomi LYWSD03MMC

V-VoRteX-V, 2023-10-30:
Thanks for the number format, I'll convert the values. But then how to correctly transfer the number to MJWSD05MMC from Home Assistant?

closed 2023-12-23 by pvvx