pvvx/ATC_MiThermometer #401: MJWSD05MMC wrong formattred 30 date character

ariekraakjr, 2023-10-30:
I love this firmware for the my bt devices! (have ten of them around the house) 
My MJWSD05MMC(the rest are the small ones) is formatting the 3 in 30 wrong at the date place(see pic)
Is there a solution for this?

[image]

pvvx, 2023-10-30:
Unfortunately, the segments of the highest digit are combined hardware on the LCD display.

```
      --6.0--         --6.4--               --7.4--         --8.4--
    |         |     |         |           |         |     |         |
   5.4       5.5   5.1       6.5      /  6.1       7.5   7.1       8.5
    |         |     |         |      /    |         |     |         |
      --5.6--         --5.2--      5.0      --6.2--         --7.2--
    |         |     |         |    /      |         |     |         |
   6.0       5.7   5.3       6.6  /      6.3       7.6   7.3       8.6
    |         |     |         |           |         |     |         |
      --6.0--         --6.7--               --7.7--         --8.7--
```
we'll have to swap the "month" and "days".
This will be fixed in version 4.5  (in an update, in an hour).

closed 2023-10-30 by pvvx

ariekraakjr, 2023-10-31:
Wow, that's fast. Thanks