pvvx/ATC_MiThermometer #402: Can't connect to a LYWSD03MMC that has the at1441 Release 78 firmware with the pvvx Flasher

jorkkiporkki, 2023-11-07:
This happens when I try to connect with the Flasher on this site:

9:26:28: Searching for devices
19:27:12: Get Advertising MAC (Web Experimental Features!)
19:27:12: Connecting to: ATC_xxxx
19:27:50: MAC: xxxxxxxxxx, Bat: 99%, Vbat: 3097 mV, Temp: 22.9°C, Humi: 50%, Count: 3
19:27:55: TypeError: getDevVersion(...).then is not a function
19:27:55: Reconnect 1 from 5
.
.
19:29:16: Disconnected.

Tried both with Android 12 device and Windows 10 / Edge with Web Experimental features enabled. Tried removing battery too.
Works with the Flasher from at1441

Keltharak, 2023-11-07:
You need to dl the firmware and use the [TlinkOTA ](https://pvvx.github.io/ATC_MiThermometer/TelinkOTA.html) flasher to upgrade first time. 
I juste did it two times few minutes ago.

jorkkiporkki, 2023-11-07:
Thanks! I had also found my courage to try it after I wrote the note. Worked here too  but maybe the compatibility could be easily improved.

pvvx, 2023-11-08:
Firmware from atc1441 does not have any identifiers that can be used to determine what kind of firmware it is.

jorkkiporkki, 2023-11-08:
Thanks for the quick response. All good now, Maybe in the future you could agree on an identifier with Aaron to make it easier.

closed 2023-11-08 by jorkkiporkki

pvvx, 2023-11-08:
They are typical for all BLE devices and are described in the standard.

UUID_MODEL_NUMBER		0x2A24 // Model Number String
UUID_SERIAL_NUMBER		0x2A25 // Serial Number String
UUID_FIRMWARE_REV		0x2A26 // Firmware Revision String
UUID_HARDWARE_REV		0x2A27 // Hardware Revision String
UUID_SOFTWARE_REV		0x2A28 // Software Revision String
UUID_MANUFACTURER_NAME	0x2A29 // Manufacturer Name String
...