pvvx/ATC_MiThermometer #403: Bricked after flashing z03mmc.bin [HELP]

QUINRY, 2023-11-11:
Hi,

İ'm flashed z03mmc.bin for zigbee support for my LYWSD03MMC.

After flash, device shows temp. and humidity but not BLE is not working.

[image]

After a few seconds, the screen stays like this and no ble appears again.

[image]

How can i fix this?

QUINRY, 2023-11-11:
@pvvx

pvvx, 2023-11-11:
Without registration in the Zigbee network, the thermometer does not work. Registration of the thermometer in the Zigbee network is carried out by closing the “reset” contacts with “gnd” for 3 seconds or when turning on the power after flashing. To save power, the registration time is limited by turning off the thermometer after 45 seconds.

QUINRY, 2023-11-11:
> Without registration in the Zigbee network, the thermometer does not work. Registration of the thermometer in the Zigbee network is carried out by closing the “reset” contacts with “gnd” for 3 [...]

Can't I flash any other firmware anymore?

pvvx, 2023-11-11:
Use Zigbee OTA.

QUINRY, 2023-11-11:
> Zigbee OTA

Where can I find it and how can I do it?

pvvx, 2023-11-11:
You have flashed a program for the thermometer to work on the Zigbee network.
The ZigBee standard has "ZigBee OTA" in its specification.

wildekek, 2023-11-14:
Your best bet is using this: https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files

closed 2023-11-21 by pvvx