pvvx/ATC_MiThermometer #408: Send version string

andreas-bulling, 2023-11-13:
With many sensors distributed across the house, it'd be super useful if the thermometer also sent the version string of the firmware so that it can be nicely seen in HomeAssistant for example.

pvvx, 2023-11-13:
This standard feature for BLE can be implemented in any device without increasing the battery consumption of the thermometer for the constant additional transmission of one-time information.

[image]

tmugan, 2023-11-14:
This sounds like a good idea to me. 
https://community.home-assistant.io/t/make-ble-softwarerevisionstring-available/640675?u=tmugan

closed 2023-12-23 by pvvx