pvvx/ATC_MiThermometer #410: Q: SJWS01LM is now supported?

ilgrank, 2023-11-15:
Hi
I see you have added SJWS01LM firmware to the list of official firmwares, and merged #310 into the repo, but SJWS01LM is not among the list of supported devices. 
..what is the current status of SJWS01LM?
Thanks!

pvvx, 2023-11-15:
There is nothing to add or correct to SJWS01LM. The official firmware works well.

ilgrank, 2023-11-16:
Well, what is missing from SJWS01LM is the ability to use custom data formats (ATC,PVVX and BHome), the possibility to alter beacon frequency and so on

pvvx, 2023-11-16:
ATC and PVVX formats are not suitable for SJWS01LM and CGPR1.
Unbinding from MiHome is done in TelinkMiFlasher.
In Home Assistant, SJWS01LM and CGPR1 will be recognized automatically.

ilgrank, 2023-11-16:
You're right about unbinding, thanks!
...but  adjustable RF TX Power and Bluetooth advertising interval frequency can't be set with stock firmware
As far as I understand, these should be possible on SJWS01LM , no?

pvvx, 2023-11-16:
RF TX boost will fail if powered by CR2032. The device will crash and restart. Higher RF TX levels require a larger power supply.
The advertising interval is set to be compatible with all BT adapters.

ilgrank, 2023-11-16:
> RF TX boost will fail if powered by CR2032
It is an issue just of SJWS01LM? I have the LYWSD03MMC  without capacitors and I have upped the TX power slightly with no issues.
As these sensors are in my basement, and I have a pretty good antenna at the receiver on the upper floor, there's not much else I can do to have better reception alas (with stock FW I can't receive the SJWS01LM beacon, while the reception is good on LYWSD03MMC  with your firmware)

I also lowered the beacon interval to 15 seconds to save battery (as per your consumption graphs) and I had hope to do the same with SJWS01LM

Ircama, 2023-11-20:
> Hi I see you have added SJWS01LM firmware to the list of official firmwares, and merged #310 into the repo, but SJWS01LM is not among the list of supported devices. ..what is the current status of SJWS01LM? Thanks!

That PR was simply referring to a preliminary support of the native SJWS01LM protocol in [atc_mi_interface](https://github.com/pvvx/ATC_MiThermometer/tree/master/python-interface), including tools and related codec and does not imply the need of a specific custom firmware. A [completion of the work](https://github.com/ircama/ATC_MiThermometer/tree/ircama-SJWS01LM) might be pulled in the near future, depending on the available time to test and close it.

> I also lowered the beacon interval to 15 seconds to save battery (as per your consumption graphs) and I had hope to do the same with SJWS01LM

The native firmware of the SJWS01LM device appears to have a beacon interval of about 10 minutes (with one repetition in the subsequent minute: each beaconing consists of up to 4 frames, generally 1 or 2, with encrypted content including the battery percentage level).

ilgrank, 2023-11-20:
Thanks for the clarification @Ircama  and @pvvx !
As a side question, is there any range improvement going from BLE (LR) to Zigbee?
I have seen plenty of BLE vs Zigbee comparisons, but no BLE LR vs Zigbee ones.

closed 2023-11-20 by ilgrank

pvvx, 2023-11-21:
[Comparison Table](https://www.bluetooth.com/blog/wireless-connectivity-options-for-iot-applications-technology-comparison/)
Real measurements on TLSR825x chips with line of sight and 0 dB transmitter power:
Zigbee - 100..500 м
LE Long Range - 200..1000 м
At the same data transmission frequency, Zigbee loses due to higher battery consumption.

aptem334, 2023-11-21:
> В будущем, если больше ничего не останется, можно преобразовать SJWS01LM в Zigbee ...
> SJWS01LM невозможно разобрать, и любая ошибка в прошивке означает, что датчик выброшен в корзину.

Добрый день, в итоге удалось преобразовать в zigbee?

evlo, 2023-11-25:
Sorry for the offtopic, but you are crazy, in the good way, i was replacing all of mine WSD500A and with the LYWSD03MMC, but now, it means I can flash them to the BLE and have sensor that does not need pairing and works of AAA and not coin cells?

evlo, 2023-11-25:
A bit offtopic, I do have quite a few WSD500A

[image]

It does say ts0201 as its zigbee ID, but it looks rather different
https://pvvx.github.io/TS0601_TZE200/
https://pvvx.github.io/TS0201_TZ3000/

[image]
but i guess it still does have GND VCC and data pins, maybe
[image]

pvvx, 2023-11-25:
@evlo - https://github.com/pvvx/BLE_THSensor/issues/3#issuecomment-1789327904

ilgrank, 2023-11-26:
@pvvx : sorry to come at the 'custom firmware' topic again, but there seems to be another issue with using the standard firmware.
[According to the OpenMQTTGateway devs,](https://github.com/1technophile/OpenMQTTGateway/issues/1815#issuecomment-1826858128): 

> The SJWS01LM seems to only be supported by decrypting the encrypted BLE advertising data, along with the individual key.
> https://custom-components.github.io/ble_monitor/by_brand#xiaomi
> Decryption is currently not yet supported by OpenMQTTGateway

..so another reason to have a custom firmware would be to have the data advertised in unencrypted format, if I understand well

pvvx, 2023-11-26:
SJWS01LM uses a primitive water sensor, low reliability. There is no point in considering such a sensor for an emergency system. This is a children's toy.

ilgrank, 2023-11-27:
Isn't a primitive sensor better than nothing? :)
Jokes apart, do you suggest any better BLE alternative? I own a couple of Shelly Flood, but they are wifi and their battery doesn't last very long alas..

pvvx, 2023-11-28:
There is no alternative to BLE.
Not just the battery. But also in terms of range in LE Long Range, as well as in the absence of a coordinator.

evlo, 2023-11-28:
Since we are suggesting, you can do diy with ie. optical flood sensor and hct-06 or similar uart transceiver and hande the raw data on the device receiver is plugged in to.

pvvx, 2023-11-29:
There has been a solution for a long time, but it requires 3 details:
1. Diode (several MHz)
2. Resistor
3. Capacitor

The program must be supplemented with a special function (developed a long time ago). The two sensor wires must be short and insulated. Sensitivity from several pF. That is, it “feels” the approach of a hand.
A complete analogue of https://github.com/rbaron/b-parasite is obtained

But each form of sensor requires adjustments due to different sizes, insulation and different parts.

Ircama, 2023-11-29:
> > The SJWS01LM seems to only be supported by decrypting the encrypted BLE advertising data, along with the individual key.

It can be decoded: you simply need the bindkey.

You can test a [preliminary version of the Python API](https://github.com/Ircama/ATC_MiThermometer/tree/ircama-SJWS01LM/python-interface), which also supports Xiaomi Mijia devices like SJWS01LM, by installing it locally.

The following program decodes and prints the first 20 BLE advertisements of one or more devices implementing the Xiaomi Mijia BLE advertisement protocol, detecting flood and battery data natively produced by the SJWS01LM sensor.

The bindkey can be obtained via [TelinkMiFlasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html).

```python
import asyncio
from bleak import BleakScanner
from bleak.uuids import normalize_uuid_str
from functools import partial
from pprint import pprint
from atc_mi_interface import mi_like_format, atc_mi_advertising_format

bindkey = {
    "54:EF:44:C6:33:6B": bytes.fromhex("ff1b17f0a386d27d407529a9f0463653"),
    "54:EF:44:AA:BB:CC": bytes.fromhex("aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa"),
    # ...
}

async def ble_coro():
    count = [0]
    MAX_COUNT = 20
    stop_event = asyncio.Event()

    def detection_callback(count, device, advertisement_data):
        format_label, adv_data = atc_mi_advertising_format(advertisement_data)
        if format_label != "mi_like" or not adv_data:
            return
        mac_address = bytes.fromhex(device.address.replace(":", ""))
        print(
            f"{count[0]}. {device}. Format: {format_label}. "
            f"RSSI: {advertisement_data.rssi}."
        )
        atc_mi_data = mi_like_format.parse(
            adv_data,
            mac_address=mac_address,
            bindkey=(
                bindkey[device.address] if device.address in bindkey else None
            )
        )
        print(" ", atc_mi_data)
        flooding = atc_mi_data.search_all("^flooding$")
        battery = atc_mi_data.search_all("^battery_level")
        if flooding:
            print()
            print("Flooding:", flooding[0])
        if battery:
            print()
            print(f"Battery: {battery[0]}%")
        print('-' * 80)
        count[0] += 1
        if count[0] == MAX_COUNT:
            stop_event.set()

    async with BleakScanner(
        detection_callback=partial(detection_callback, count)
    ) as scanner:
        await stop_event.wait()
    print("Stopped")

asyncio.run(ble_coro())
```