pvvx/ATC_MiThermometer #411: Bricked MJW5D05MMMC after flashing to fw ver 4.5 using Telink flasher

pololevu, 2023-11-15:
I just updated a couple of LYWSD03MMC) devices to v4.5 firmware which went well (not my first time). I then tried to update a MJWSD05MMC from v4.4. All seemed to go well but after supposedly successful completion, the device stayed on the flashing '- - -' screen and was unresponsive to button presses. I tried removing the battery but the device seems dead, no display at all. Can you point me at instruction on how to reflash it?

Thanks.

log:
23:27:11: Connecting to: BTH_B92DA2
23:27:15: Hardware Revision String: V2.3
23:27:15: Software Revision String: V4.4
23:27:15: Firmware Revision String: github.com/pvvx
23:27:15: Detected custom Firmware
23:27:15: Hardware Version: MJWSD05MMC V2.3, Software Version: 4.4, Sensor: SHT4x
23:27:15: Custom config HEX string: 55000200005004ae3100095a
23:27:27: Start DFU
23:28:23: Update done after 55.132 seconds
23:28:27: Disconnected.
23:31:26: Reconnect
23:31:26: NetworkError: Bluetooth Device is no longer in range.
23:31:26: Reconnect 1 from 5
23:31:26: NetworkError: Bluetooth Device is no longer in range.
23:31:26: Reconnect 2 from 5
23:31:26: NetworkError: Bluetooth Device is no longer in range.
23:31:26: Reconnect 3 from 5
23:31:26: NetworkError: Bluetooth Device is no longer in range.
23:31:26: Reconnect 4 from 5
23:31:26: NetworkError: Bluetooth Device is no longer in range.
23:31:26: Reconnect 5 from 5
23:31:26: NetworkError: Bluetooth Device is no longer in range.
23:31:26: Something went wrong, too many reconnects
23:31:54: Searching for devices
23:32:37: NotFoundError: User cancelled the requestDevice() chooser.

pvvx, 2023-11-16:
They changed something in the Bluetooth Web API in Chrome on Windows... I haven’t figured it out yet. There are constant connection requests from the adapter without an actual connection to the MJWSD05MMC.
I used Android and everything goes well.

pvvx, 2023-11-16:
I figured it out - it’s not the API that’s to blame, but the USB-BT adapter on the ATS2851 chip.
There is a mistake in it. Causes a cyclic connection... 
I used another Bluetooth adapter, the built-in RZ616, and everything was fine.

pololevu, 2023-11-16:
Thanks for the fast response, FWIW I was using Linux with an AX200 WiFi/Bluetooth adapter, I've used it before and generally it works quite well. I had just flashed two smaller devices with it before trying to flash the MJW5D05MMMC. 

Now I have a device which has a completely blank screen and unsurprisingly doesn't show up on a bluetooth scan nor respond to button presses. I would like to try to re-flash it over a serial connection if possible, I should be able to solder wires on to it (provided the pads are not super tiny) but I don't know where to connect to or what serial interface (eg signalling level) or flashing software to use.

Any pointers or suggestions welcome.

pvvx, 2023-11-16:
On the "[The USB-COM adapter writes the firmware in explorer. Web version](https://github.com/pvvx/ATC_MiThermometer#the-usb-com-adapter-writes-the-firmware-in-explorer-web-version)" home page.

> Now I have a device which has a completely blank screen and unsurprisingly doesn't show up on a bluetooth scan nor respond to button presses.

I can't reproduce a similar situation.
The reason is not clear.

pololevu, 2023-11-16:
Thanks again for the response. I think my issue may have been a stupid mistake on my part - as I mentioned I had just re-flashed two LYWSD03MMC devices successfully, then paired the MJW5D05MMMC and just hit the button to flash it without checking the firmware version so I suspect I accidentally flashed it with the wrong software hence bricking it. I don't think there was a problem with the flasher or the bluetooth interface, just pure stupidity on my part. I will try to update the device using a USB-COM adapter and update and/or close this after that.

pvvx, 2023-11-16:
Firmware from LYWSD03MMC should work on MJWSD05MMC. That is, BLE will work and nothing more.

pololevu, 2023-11-17:
Awesome! I can see the device and as you say the BLE is working. I can connect to it from my phone using Telink or TelinkOTA but when I select the correct firmware and try to down load it the message says it is trying to send block 7, it pauses then prints the message "Update error: NotSupportedError: GATT Error Unknown" and then disconnects. I am trying to flash the BTH_v45.bin downloaded from the Github link.

closed 2023-11-17 by pololevu