pvvx/ATC_MiThermometer #418: Strange repeated temperature readings in v4.5

chatziko, 2023-11-24:
Hello, I just upgraded two sensors to v4.5 (from v4.3 if I recall correctly) and started seeing strange temperature readings in both of them.

Below you can see the difference before and after the upgrade (which happened around 10:00 yesterday):

[image]

Zooming in around 21:00 (after the upgrade) we see that the sensor reports a strange value of 19.67 once per minute or so, independentely from whether the actual value was below or above 19.67:

[image]

The second sensor has exactly the same behaviour, but its frequently reported value is 20.52. I tried moving it to a colder location to see that happens, and the 20.52 is still reported, although much less frequently, as can be seen below.

[image]

Any idea what could be the cause of this?

pvvx, 2023-11-24:
https://github.com/pvvx/ATC_MiThermometer/issues/405

closed 2023-11-24 by pvvx

chatziko, 2023-11-24:
Oh, thanks! I would have sweared I checked the issues before posting, but I guessed I looked only at the open ones :smile:

pvvx, 2023-11-24:
The reason is not the thermometer or your actions.
The reason is Bluez and the Linux kernel.