pvvx/ATC_MiThermometer #419: Not found Telink UUID service!

txitxo0, 2023-11-26:
Hi there, 

I am trying to move fordward to the ATC4.5 to get zigbee enabled. 

I have a few xiaomi thermometers with an atc custom firmware. I am trying to go into this new version but as soon as the we tries to pair with any of my devices it says:
```
11:35:03: Searching for devices
11:35:06: Connecting to: ATC_162894
11:35:09: Not found Telink UUID service!
11:35:09: Disconnected.
```

Trying to get further, I realize in this other project the device gets to be paired https://pvvx.github.io/ATC_MiThermometer/TelinkOTA.html
```
11:40:10: Connecting to: ATC_162894
11:40:16: Connected.
```
I don't know if it is possible to upload the ATC46 firm from this TelinkOTA project to my device.

Only to be completely sure which device I have, I opened it to be sure is the LYWSD03MMC
[image]

Do you know what can be happening?

pvvx, 2023-11-27:
Use https://atc1441.github.io/TelinkFlasher.html
file https://github.com/pvvx/ATC_MiThermometer/raw/master/ATC_v46.bin

pvvx, 2023-11-27:
> File was selected, size: 82612 bytes

You have flashed the BLE firmware. Use [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html).

txitxo0, 2023-11-27:
Ohhh! I got it now! Thank you! Firstly update with the other tool and then with the most updated one.

Big thanks!

Misiu, 2023-11-30:
I have a similar issue, with the same error message. I didn't want to create a new issue, so I'll describe the problem here.
I've bought a new LYWSD03MMC, I was able to flash it with 4.5 firmware and set BTHome v2 advertise and change the device name. The device was discovered by Home Assistant.

Now I'm trying to change some settings, but everytime I try to connect I get error:
```Not found Telink UUID service!```

I've tried using:
https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html

Log:
```
15:17:53: Searching for devices
15:18:08: Connecting to: ATC_BIURO
15:18:46: Disconnected.
15:18:46: Not found Telink UUID service!
```

https://atc1441.github.io/TelinkFlasher.html
Log:
```
15:04:43: Searching for devices
15:05:00: Connecting to: ATC_BIURO
15:05:01: Disconnected.
15:05:01: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
15:05:01: Reconnect 1 from 5
15:05:08: NetworkError: Connection Error: Connection attempt failed.
15:05:08: Reconnect 2 from 5
15:05:49: Disconnected.
15:05:49: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
15:05:49: Reconnect 3 from 5
15:05:51: Disconnected.
15:05:51: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
15:05:51: Reconnect 4 from 5
15:05:59: NetworkError: Connection Error: Connection attempt failed.
15:05:59: Reconnect 5 from 5
15:06:37: Disconnected.
15:06:37: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
15:06:37: Something went wrong, to many reconnect's
```

https://pvvx.github.io/ATC_MiThermometer/TelinkOTA.html
Log:
```
15:07:56: Searching for devices
15:08:03: NotFoundError: User cancelled the requestDevice() chooser.
15:08:03: Reconnect 1 from 5
15:08:44: Searching for devices
15:09:48: Connecting to: ATC_BIURO
15:10:31: Disconnected.
15:10:31: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
15:10:31: Reconnect 1 from 5
15:10:33: Disconnected.
15:10:33: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
15:10:33: Reconnect 2 from 5
15:10:41: NetworkError: Connection Error: Connection attempt failed.
15:10:41: Reconnect 3 from 5
15:11:27: Disconnected.
15:11:27: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
15:11:27: Reconnect 4 from 5
15:12:05: Disconnected.
15:12:05: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
15:12:05: Reconnect 5 from 5
15:12:13: NetworkError: Connection Error: Connection attempt failed.
15:12:13: Something went wrong, to many reconnect's
```

Funny thing is that Home Assistant detects the device correctly even after restart (I didn't add the device to Home Assistant, but it gets discovered after HA restart)

I'm not sure what to do with the device to get the connection working. I've restarted my PC a couple of times and I still get the same error over and over.
Can I reset the device somehow? Thans for the advice

pvvx, 2023-12-01:
This is a problem with your BT adapter in Windows. Some adapters work at intervals of up to 10 seconds...