pvvx/ATC_MiThermometer #422: Cannot connect to ATC_xxxxx

phellarv, 2023-11-28:


phellarv, 2023-11-28:
I have now tried 2 different PC's. Same result.
[image]
And there it just hangs.

dgoadby, 2023-11-28:
Is this your first time trying to connect with BLE or do you already have working devices?

If this is your first time then your problem might be the Bluetooth hardware. i had laptops with Bluetooth and thay didn't seem to work and I was pulling my hair out until I bought a recommended BLE dongle and, since then, all has been well. The dongle is: tp-link Bluetooth 5.0 Nano USB adaptor.

pvvx, 2023-11-28:
Regarding the operation of the ATC_ota_40000.bin firmware, please contact https://github.com/devbis/z03mmc.

phellarv, 2023-11-28:
I tried everything on 2 different Linux PC's, but when I booted into windows it worked right away. Painful, but if it works, it works.

closed 2023-11-28 by phellarv

pvvx, 2023-11-28:
Linux only works with older versions of Bluetooth released before 2016.

phellarv, 2023-11-28:
It did work with 3 out of 4 devices which I flashed today, so it is not consistent. All bought from the same vendor at the same time. And they had the same firmware before I flashed them.
But noted. I have a drive with windows on which I can boot when needed.