pvvx/ATC_MiThermometer #425: QUESTION: Does MJWSD05MMC support both MiHome app and XiaomiGateway 3 in HA?

JonathanKang, 2023-12-01:
Does MJWSD05MMC support both MiHome app and XiaomiGateway 3 in HA with this custom firmware, as MJWSD03MMC does?

pvvx, 2023-12-01:
The Mi Home user agreement prohibits such actions. You cannot use Xiaomi Clouds and their software without violating the user agreement.

JonathanKang, 2023-12-01:
> Publish the remaining parts in open sources to implement compatibility with MiHome - then it will be possible to add them to alternative firmware without breaking laws.

I'm using your custom firmware in my LYWSD03MMC thermometer, both in MiHome app and XiaomiGateway 3 in HA. It works as expected. Does this also violates the user agreement? Can MJWSD05MMC work similarly with LYWSD03MMC? I tried this earlier today, but it doesn't seem to work.

pvvx, 2023-12-01:
> Does this also violates the user agreement? 

Have you read it? The text appears when the device is connected and in the menu. :)

JonathanKang, 2023-12-01:
> Have you read it? The text appears when the device is connected and in the menu. :)

Turns out it does violate the user agreement. 

Well, let's put the user agreement aside for now, is it possible to work work MJWSD05MMC with 'Xiaomi Gateway 3'?

pvvx, 2023-12-01:
> Well, let's put the user agreement aside for now, is it possible to work work MJWSD05MMC with 'Xiaomi Gateway 3'?

Where should I put it?

MJWSD05MMC seems to be made for the Chinese domestic market.
Register there.

JonathanKang, 2023-12-01:
Nice. Thanks. I'll stick with the original firmware.

closed 2023-12-01 by JonathanKang

pvvx, 2023-12-01:
Oh, another one ran away and didn’t give everyone instructions on how to attach the thermometer to MiHome and who they could then link to :)