pvvx/ATC_MiThermometer #430: Using 4.5 firmware and values from two are getting unavailable

thehijacker, 2023-12-05:
I have an odd problem. Nine LYWSD03MMC sensors with latest 4.5 firmware on them. 4 are on top floor and 5 in ground floor of the house. I also have two ESP32 flashed with Bluetooh Proxy. One for each floor (in middle of hallways). Default configuration for them:

```
esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true
```

All nine sensors have default configuration. I only changed the protocol to BTHome v2. Home assistant picked them all up automatically under BTHome integration up to 60 seconds after I changed the protocl and saved configuration. And it started collecting data.

But two of the (on top floor) are giving me issues. There are gaps between collection and Home Assistant will show unavailable status for them when he doesn't get regular updates.

Here are all with current values:

[image]

And graph for one without issues:

[image]

And graphs for the two problematic ones:

[image]

[image]

I did change the batteries on both problematic sensors and it is the same with new batteries. 

How to debug this? Bluetooth Proxy is not picking them up? Only one proxy would be enough or I need more then two? I need different ESPHome Bluetooth Proxy configuration or maybe I need to change setting on the temperature sensors?

Thank you for any suggestions.

pvvx, 2023-12-05:
I have a lot of BLE sensors and when testing ESPHome Bluetooth Proxy did not cope.
But the simplest CSR8510A10 USB-BT adapter works stably with 30..50 BLE devices.
https://github.com/pvvx/ATC_MiThermometer/discussions/328

thehijacker, 2023-12-05:
Thanks for the prompt reply. But this adapter requires PC? Will one of this adapters be able to reach all sensors in house (both floors)? The HA servers is not in ideal location (ex. center of the house).

pvvx, 2023-12-05:
It is possible to switch the thermometer to work on "LE Long Range".
The communication range in the “LE Long Range” mode exceeds the communication range of household Wi-Fi.
To compensate for battery consumption, increase the advertising interval in TelinkMiFlasher.html by 5..6 seconds.

Then you will need an RTL8761BU type adapter. How to enable BLE processing mode is described in https://github.com/pvvx/ATC_MiThermometer/issues/297.

thehijacker, 2023-12-06:
I've rebooted both ESP proxy boards and set interval to 5 seconds. It seems to fix it:

[image]

thehijacker, 2023-12-06:
Today I have received Orange Pi Zero 3 board to use as cheap backup for some service main server runs. It also has Bluetooth 5.0. Can I use it as BLE receiver?

http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-Zero-3.html

pvvx, 2023-12-07:
> Today I have received Orange Pi Zero 3 board to use as cheap backup for some service main server runs. It also has Bluetooth 5.0. Can I use it as BLE receiver?

No technical specifications. Only advertising about BT 5.0.
There is no support for Bluetooth 5.0 in the system and program.
Built-in BT chips in different RPIs usually have poor reception, low performance and instability.

mtnrun, 2023-12-09:
Apologies for these basic questions but I'm new to this and I'm having the same issue (dropped sensor readings) for which I haven't been able to find any definitive resources on the best esphome configuration if using bluetooth_proxy or a different setup altogether to get `LYWSD03MMC` sensor data reliably in home assistant.

For esphome, do you need to set `esp32_ble_tracker` and /or `bluetooth_proxy` to active? What's the best config that works with `LYWSD03MMC`? Additionally, what are the ideal configs for `LYWSD03MMC` around setting different intervals to maybe reduce conflicts with multiple units (I have 5 sensors currently)?

It seems like pvvx is suggesting that esp32 boards that act as bluetooth_proxies are always going to be brittle. Am I understanding this correctly that buying a bluetooth dongle and connecting it directly to the server running home assistant will yield better results? The part I find confusing with that suggestion is driver support. I got a beelink mini s12 and followed the suggestion to install proxmox, and then run home assistant in a VM. Are you familiar if bluetooth support for a dongle would work with such a setup? Appreciate any additional guidance.

pvvx, 2023-12-09:
> What's the best config that works with LYWSD03MMC?

The [USB-BT adapter on the CSR8510A10](https://www.home-assistant.io/integrations/bluetooth/#known-working-high-performance-adapters) works with any thermometer interval settings.
But "LE Long Range" doesn't work - the CSR8510A10 is a chip with preliminary support for Bluetooth 4.2.

pvvx, 2023-12-09:
Devices such as smart plugs, TVs, computers, smartphones and other devices with Bluetooth broadcast BLE/MESH advertisements to women within 50-100 ms.
[image]
All this interferes with reception and the adapter must accept everything.

mtnrun, 2023-12-09:
Thanks, I think I'll buy one of the CSR8510A10 adapters from the HA page you linked and see if I can get it to work. All my sensors are within 6 meter range.

pvvx, 2023-12-09:
### **A set of 2 adapters, providing all the options possible today.**

[image]

**RTL8761BU** (Best receiver sensitivity) 
1. Software loads when Linux starts (BT5.2)
2. Low speed
3. Support all PHY - Working in LE Long Range (500..1500 m)
[image]
https://aliexpress.com/item/4001097686145.html
[image]

**CSR8510A10** (Speed and large number of accepted Bluetooth 4.2 advertising packets.)
https://aliexpress.com/item/4001097686145.html
[image]

thehijacker, 2023-12-09:
If I buy one of this and connect it to a PC that is not running Home Assistant directly (RPi, etc), what would be best software to run on it, that would collect the data and pass it to Home Assistant?

I would prefer that Home Assistant automatically detects the sensors if possible. But I do not mind to add the manually as I have only 9.

pvvx, 2023-12-09:
I have not done such research. For me, to cover my private territory of hundreds of meters, one single-board computer with two specified cheap adapters is enough.

thehijacker, 2023-12-12:
Indeed it was a race condition. After one day those two sensor readings were again unreliable. But today I have received the USB BT adapter (Ugreen):

```
Bus 001 Device 021: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
[3018415.534985] usb 1-3: new full-speed USB device number 21 using xhci_hcd
[3018415.792501] usb 1-3: New USB device found, idVendor=0a12, idProduct=0001, bcdDevice=88.91
[3018415.792511] usb 1-3: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[3018415.792515] usb 1-3: Product: CSR8510 A10
[3018415.799454] Bluetooth: hci1: CSR: Setting up dongle with HCI ver=6 rev=22bb; LMP ver=6 subver=22bb; manufacturer=10
```

I have connected it now to Home Assistant server and it was detected by Bluetooth integration. So I disabled both ESP32 proxies but noew I no longer receive any readings over BTHome integration.

I am sure I need to configure this differently now. Can you point me in proper direction? Can I still use BTHome as protocol but collect the data over Bluetooth integration with this CSR8510 A10 adapter?

pvvx, 2023-12-13:
To work in BTHome in “passive mode” you need to set the following settings on the thermometer:

[image]

And also change the settings for Bluez.

* activate experimental mode for bluetooth service :
$ sudo nano /lib/systemd/system/bluetooth.service

Add --experimental at the end of the following line :
ExecStart=/usr/local/libexec/bluetooth/bluetoothd --experimental

pvvx, 2023-12-13:
Passive BLE Monitor uses Black and Bluez. For this reason, it cannot work with BT5.0.

> Indeed it was a race condition. After one day those two sensor readings were again unreliable.

He gave a comparison: [ESP versus any BT adapter.](https://github.com/pvvx/ATC_MiThermometer/discussions/328#discussioncomment-7838023)

closed 2023-12-23 by pvvx