pvvx/ATC_MiThermometer #431: Not found Telink UUID service!

sander816, 2023-12-08:
i am getting this with my sensors

10:42:57: Searching for devices
10:43:02: Connecting to: ATC_F9DEA5
10:43:06: Not found Telink UUID service!
10:43:06: Disconnected.

any idea?

pvvx, 2023-12-08:
Clear your Chrome cache or go to incognito mode.
See what happens to BLE devices in Chrome by inserting the "chrome://bluetooth-internals/".

sander816, 2023-12-08:
no difference

this is from chrome://bluetooth-internals/
Name:
ATC
Address:
A4:C1:38:F9:DE:A5
GATT Connected:
Connected
Latest RSSI:
-60
Services:
00001800-0000-1000-8000-00805f9b34fb, 00001801-0000-1000-8000-00805f9b34fb, 0000180f-0000-1000-8000-00805f9b34fb, 0000181a-0000-1000-8000-00805f9b34fb, 00001f10-0000-1000-8000-00805f9b34fb, 0000fe95-0000-1000-8000-00805f9b34fb, 00010203-0405-0607-0809-0a0b0c0d1912
Manufacturer Data:

pvvx, 2023-12-08:
It's hard for me to find this inconsistency.
Since after clearing the BLE cache in Chrome, it no longer happened to me. Before this, it rarely happened and was associated with an internal GATT error in Chrome, with the message “disconnect”.

Rewrote part of the code, check https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html v7.2

pvvx, 2023-12-08:
For Atc1441 the correct answer is: _Detected alternative Firmware? Use: https://pvvx.github.io/ATC_MiThermometer/TelinkOTA.html_

The Atc1441 firmware does not have any identifiers, and it cannot be distinguished from other DIY firmware made using examples from the Telink SDK.

sander816, 2023-12-08:
Get the same msg there

Log:
13:58:23: Searching for devices
13:58:27: Connecting to: ATC_067765
13:58:27: Detected custom Firmware

pvvx, 2023-12-08:
Сheck the battery.

sander816, 2023-12-08:
i dont know what you mean by check battery

i replaced the battery for a new one at both sensors, no difference

closed 2023-12-23 by pvvx