pvvx/ATC_MiThermometer #432: Zigbee firmware is more than 131072 bytes

jjvelar, 2023-12-10:
Hi.
Just updated to 4.5 firmware but when tried to update do Zigbee firmware Z03MMC_v0116, get error "firmware is more than 131072 bytes".
Please, advise.
Thanks in advance for your help

pvvx, 2023-12-10:
Only BLE version from 4.6 has Zigbee download support.

jjvelar, 2023-12-10:
Oh! Didn't know that. Will update to 4.6 beta and then will try to move to the Zigbee firmware.
By the way, thanks for pointing me out on how to downgrade. Have been able to do it and now I am trying to move to Zigbee!

pvvx, 2023-12-10:
More precisely, version 4.6 and above can load large files or the “*.zigbee” file format.
At the moment, "Help" has not yet been rewritten.

jjvelar, 2023-12-10:
Works like a charm. Thanks!
Quick question: once the zigbee firmware is installed, how can I update it? is it possible to do it via Zigbee2MQTT?
Thanks again!

pvvx, 2023-12-10:
I don't use Zigbee2MQTT and don't know what needs to be entered there. But everything is the same as on https://github.com/devbis/z03mmc

jjvelar, 2023-12-10:
Thanks a lot Victor @pvvx

pvvx, 2023-12-10:
It’s better to write about Zigbee there https://github.com/pvvx/ZigbeeTLc