pvvx/ATC_MiThermometer #435: Recording averaging measurements to flash memory limit

AleXSR700, 2023-12-13:
Good morning
I could not find information on this but I found that maximum interval for `Recording averaging measurements to flash memory` seems to be around 240.
Maybe it is not the actual value but the resulting interval (now at 120 min).

Could this also be disabled? E.g. I personally never look at the history. All recordings are done on my server.

Thank you for your great work!

Alex

pvvx, 2023-12-13:
[image]

AleXSR700, 2023-12-13:
Oh, sorry, I am using my phone so I can get closer to the sensors (rather than moving them or my notebook ;) ).

I was thinking of battery and flash life span. Both probably less problematic, but since it is redundant for a lot of people, I figured "every little helps" :)

pvvx, 2023-12-13:
Updating BLE firmware on a regular adapter takes up to 40 seconds. During the update, the average current from the thermometer's battery is about 1 mA. This is equal to 1x40/(60x60) = 0.01111 mAh