pvvx/ATC_MiThermometer #436: Phantom sensors in BTHomev2 mode

Komzpa, 2023-12-13:
Hi,

I have one device that shows unusual properties: it grows itself extra reported sensors in BTHome mode. The thermometer is put into the heated floor so regularly heated 30-35C, and in Home Assistant it already reported that it had three batteries, light sensor, moisture sensor, cold sensor (while heated), pressure sensor. 

[image]

It was using firmware 4.5, I just upgraded it to 4.6 and continuing observations. Let me know if I can report something useful here.

Log from upgrade:
```
5:21:46 PM: Searching for devices
5:21:51 PM: Connecting to: ATC_F8609B
5:21:56 PM: Hardware Revision String: B1.6
5:21:56 PM: Software Revision String: V4.5
5:21:56 PM: Firmware Revision String: github.com/pvvx
5:21:56 PM: Detected custom Firmware
5:21:56 PM: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.5, Sensor: SHT4x
5:21:56 PM: Custom config HEX string: 55871000002804a9313104b4
5:21:59 PM: Load firmware file 'ATC_v46.bin'...
5:21:59 PM: File: ATC_v46.bin
5:21:59 PM: File size: 82612 bytes
5:21:59 PM: Count: 5164
5:22:01 PM: Start DFU
5:22:56 PM: Update done after 53.619 seconds
5:23:00 PM: Disconnected.
5:23:04 PM: Searching for devices
5:23:06 PM: Connecting to: ATC_F8609B
5:23:11 PM: Hardware Revision String: B1.6
5:23:11 PM: Software Revision String: V4.6
5:23:11 PM: Firmware Revision String: github.com/pvvx
5:23:11 PM: Detected custom Firmware
5:23:11 PM: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
5:23:11 PM: Custom config HEX string: 55871000002804a9313104b4
5:24:32 PM: New custom config: [135, 16, 0, 0, 40, 4, 174, 49, 100, 4, 180]
5:24:32 PM: Settings 55871000002804AE316404B4 was send successful
5:24:33 PM: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
5:24:33 PM: Custom config HEX string: 55871000002804ae316404b400
5:25:07 PM: Disconnected.
```

Komzpa, 2023-12-13:
The BT adapter is Raspberry Pi 3 built-in one:
[image]

pvvx, 2023-12-13:
> BT adapter is Raspberry Pi 3 built-in

This is what causes all your problems.

closed 2023-12-23 by pvvx

Komzpa, 2023-12-24:
If someone else will experience a similar issue: workaround is to enable encryption of beacon, then radio issues will cause decryption to fail and skip the measurement rather than getting broken data into the system.

Ernst79, 2024-03-20:
BTHome integration has been adjusted, such that it only asks to reauth after two failing messages (right after eachother).