pvvx/ATC_MiThermometer #437: Question: Why zigbee?

hapklaar, 2023-12-13:
My HASS supports both BLE and Zigbee. Is there any reason to convert a LYWSD03MMC to Zigbee firmware? I read it consumes a bit more energy and was not able to discover any upsides.

Who can educate me?

pvvx, 2023-12-13:
There is no point in switching to Zigbee.

1. BLE in version BT4.2 has a communication range with a conventional adapter of about 50..70 meters in a straight line.
2. Zigbee option, if the coordinator uses a new modern chip - 200..300 meters in a straight line.
3. BLE in the LE Long Range version has a communication range of 500 meters in a straight line with a conventional adapter.

"LE Long Range" can be enabled in any provided firmware.

hapklaar, 2023-12-13:
Thank you, I thought as much.

About the Long Range mode for BLE. Do I need a special adapter for that or will any Bluetooth 5.x adapter do?

pvvx, 2023-12-13:
Button thermometers switch to "legacy" mode for connection at the touch of a button.
Tested USB-BT adapter ($2..3) - all with RTL8761BU chips.

closed 2023-12-13 by hapklaar