pvvx/ATC_MiThermometer #439: LYWSD03MMC can't flash ZigBee Firmware

MikealSouth, 2023-12-15:
I've tried to flash the Zigbee Firmware to two different LYWSD03MMC Temperature and Humidity Sensors OTA via https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html

So I've upgraded to Version 4.6 and then tried to flash the ZigBee Firmware.
It stops after the message "Clear Ext.OTA region (140kib)..." then the displays shows three "_" symbols as pictured here.
This will continue until I remove and reset the battery or until I click "Reconnect" on the flashing website. Picture is attached.
[image]

Could someone please point me to what I am doing wrong?

Log
```
18:56:31: Searching for devices 
18:58:45: Connecting to: ATC_735252
18:59:05: Reconnect
18:59:12: Hardware Revision String: B1.6
18:59:12: Firmware Revision String: github.com/pvvx
18:59:12: Detected custom Firmware
18:59:12: Hardware Revision String: B1.6
18:59:12: Software Revision String: V4.6
18:59:12: Firmware Revision String: github.com/pvvx
18:59:12: Detected custom Firmware
18:59:12: Detected alternative Firmware? Use: https://pvvx.github.io/ATC_MiThermometer/TelinkOTA.html
18:59:12: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
18:59:12: Custom config HEX string: 55830000002804a9313104b4
18:59:12: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
18:59:12: Custom config HEX string: 55830000002804a9313104b4
18:59:38: Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/Z03MMC_v0117.bin'...
18:59:38: File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/Z03MMC_v0117.bin
18:59:38: File size: 143844 bytes
18:59:38: Count: 8991
18:59:43: Start DFU
18:59:45: Clear Ext.OTA region (140kib)...
19:02:55: Disconnected.
19:02:55: Reconnect
19:02:55: Disconnected.
19:03:10: Reconnect
19:03:16: Searching for devices
19:03:42: Connecting to: ATC_76A52B
19:03:55: Hardware Revision String: B1.6
19:03:55: Software Revision String: V4.5
19:03:55: Firmware Revision String: github.com/pvvx
19:03:55: Detected custom Firmware
19:03:55: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.5, Sensor: SHT4x
19:03:55: Custom config HEX string: 55871000002804a9313104b4
19:04:00: Load firmware file 'ATC_v46.bin'...
19:04:00: File: ATC_v46.bin
19:04:00: File size: 82612 bytes
19:04:00: Count: 5164
19:04:04: Start DFU
19:04:28: Update done after 22.941 seconds
19:04:28: Disconnected.
19:06:33: Reconnect
19:06:34: Hardware Revision String: B1.6
19:06:34: Software Revision String: V4.6
19:06:34: Firmware Revision String: github.com/pvvx
19:06:34: Detected custom Firmware
19:06:34: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
19:06:34: Custom config HEX string: 55871000002804a9313104b4
19:06:46: Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/Z03MMC_v0117.bin'...
19:06:46: File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/Z03MMC_v0117.bin
19:06:46: File size: 143844 bytes
19:06:46: Count: 8991
19:06:50: Start DFU
19:06:52: Clear Ext.OTA region (140kib)...
19:13:04: Disconnected.
19:13:04: Reconnect
```

pvvx, 2023-12-15:
1. Try doing this:
[image]
2. Disconnect
3. Reconnect
4. OTA.

MikealSouth, 2023-12-15:
Hey pvvx, thank you for the mega fast response, I've tried the step you've mentioned several times now, unfortunately nothing has changed.
Then I've changed the BT Advertising type to "PVVX", screenshot attached
[image]
The BT Adapter used is a Apple MacBook M1 from 2020

Log:
```
21:06:57: Hardware Revision String: B1.6
21:06:57: Software Revision String: V4.6
21:06:57: Firmware Revision String: github.com/pvvx
21:06:57: Detected custom Firmware
21:06:57: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
21:06:57: Custom config HEX string: 55871000002804a9313104b4
21:07:08: Connecting to: ATC_735252
21:07:13: Hardware Revision String: B1.6
21:07:13: Software Revision String: V4.6
21:07:13: Firmware Revision String: github.com/pvvx
21:07:13: Detected custom Firmware
21:07:13: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
21:07:13: Custom config HEX string: 55830000002804a9313104b4
21:07:21: Settings 55 was send successful
21:07:21: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
21:07:21: Custom config HEX string: 55830000002804a9313104b400
21:07:30: Disconnected.
21:07:32: Reconnect
21:07:34: Hardware Revision String: B1.6
21:07:34: Software Revision String: V4.6
21:07:34: Firmware Revision String: github.com/pvvx
21:07:34: Detected custom Firmware
21:07:34: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
21:07:34: Custom config HEX string: 55830000002804a9313104b4
21:07:39: Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/Z03MMC_v0117.bin'...
21:07:39: File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/Z03MMC_v0117.bin
21:07:39: File size: 143844 bytes
21:07:39: Count: 8991
21:07:41: Start DFU
21:07:43: Clear Ext.OTA region (140kib)...
21:10:18: Disconnected.
21:10:18: Reconnect
21:10:18: Disconnected.
21:10:30: Reconnect
21:10:34: Reconnect
21:11:02: Searching for devices
21:11:14: Connecting to: ATC_735252
21:11:17: Hardware Revision String: B1.6
21:11:17: Software Revision String: V4.6
21:11:17: Firmware Revision String: github.com/pvvx
21:11:17: Detected custom Firmware
21:11:17: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
21:11:17: Custom config HEX string: 55830000002804a9313104b4
21:11:22: Settings 55 was send successful
21:11:22: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
21:11:22: Custom config HEX string: 55830000002804a9313104b400
21:11:29: Disconnected.
21:11:30: Reconnect
21:11:37: Hardware Revision String: B1.6
21:11:37: Software Revision String: V4.6
21:11:37: Firmware Revision String: github.com/pvvx
21:11:37: Detected custom Firmware
21:11:37: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
21:11:37: Custom config HEX string: 55830000002804a9313104b4
21:11:43: Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/Z03MMC_v0117.bin'...
21:11:43: File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/Z03MMC_v0117.bin
21:11:43: File size: 143844 bytes
21:11:43: Count: 8991
21:11:46: Start DFU
21:11:48: Clear Ext.OTA region (140kib)...
21:14:14: New custom config: [129, 0, 0, 0, 40, 4, 169, 49, 49, 4, 180]
21:14:14: Settings 55810000002804A9313104B4 was send successful
21:14:14: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
21:14:14: Custom config HEX string: 55810000002804a9313104b400
21:14:17: Disconnected.
21:14:17: Reconnect
21:14:17: Disconnected.
21:14:38: Searching for devices
21:14:45: Connecting to: ATC_735252
21:14:48: Hardware Revision String: B1.6
21:14:48: Software Revision String: V4.6
21:14:48: Firmware Revision String: github.com/pvvx
21:14:48: Detected custom Firmware
21:14:48: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
21:14:48: Custom config HEX string: 55810000002804a9313104b4
21:14:53: New custom config: [129, 0, 0, 0, 40, 4, 169, 49, 49, 4, 180]
21:14:53: Settings 55810000002804A9313104B4 was send successful
21:14:53: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
21:14:53: Custom config HEX string: 55810000002804a9313104b400
21:15:12: Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/Z03MMC_v0117.bin'...
21:15:12: File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/Z03MMC_v0117.bin
21:15:12: File size: 143844 bytes
21:15:12: Count: 8991
21:15:14: Start DFU
21:15:16: Clear Ext.OTA region (140kib)...
21:19:12: Disconnected.
21:19:12: Reconnect
```

pvvx, 2023-12-15:
I don't have V1.6. I can only imitate on a breadboard.
```
23:24:01: Searching for devices
23:24:10: Connecting to: ATC_119EFB
23:24:14: Hardware Revision String: B1.6
23:24:14: Software Revision String: V4.6
23:24:14: Firmware Revision String: github.com/pvvx
23:24:14: Detected custom Firmware
23:24:14: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
23:24:14: Custom config HEX string: 55859000002804a9313104b4
23:24:17: Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/Z03MMC_v0117.bin'...
23:24:17: File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/Z03MMC_v0117.bin
23:24:17: File size: 143844 bytes
23:24:17: Count: 8991
23:24:19: Start DFU
23:24:20: Clear Ext.OTA region (141kib)...
23:24:27: Clean Flash OTA region: 0x040000, size: 144 kbytes. Go OTA...
23:25:13: Update done after 45.388 seconds
23:25:17: Disconnected.
```

MikealSouth, 2023-12-15:
I have several spares, if you're interested to I am more then happy to send you one. Pretty sure that I am not the only person interested in this.
Please just send me a shipping address to pvvx@whisperguard.maskmy.id

pvvx, 2023-12-15:
> 21:14:53: Settings 55810000002804A9313104B4 was send successful

You do not enable "Screen off:".

MikealSouth, 2023-12-15:
I am sorry but this didn't help.
Pretty sure that you're looking for `Settings 55 was send successful`, that has been set now.
So I've tried it twice and nothing changed, then I removed the battery and reinserted it, the screen did not come back on but I could connect to the thermometer.
So that worked.
Tried to flash again, waited for three minutes.
Nothing, stuck at `Clear Ext.OTA region (141kib)...`

Is there anything else I can try?
What BT adapter are you using?
```
22:35:20: Reconnect
22:35:34: Searching for devices
22:35:47: Connecting to: ATC_735252
22:35:57: Hardware Revision String: B1.6
22:35:57: Software Revision String: V4.6
22:35:57: Firmware Revision String: github.com/pvvx
22:35:57: Detected custom Firmware
22:35:58: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
22:35:58: Custom config HEX string: 55810000002804a9313104b4
22:36:02: Settings 55 was send successful
22:36:02: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
22:36:02: Custom config HEX string: 55810000002804a9313104b400
22:36:13: New custom config: [129, 128, 0, 0, 40, 4, 169, 49, 49, 4, 180]
22:36:13: Settings 55818000002804A9313104B4 was send successful
22:36:13: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
22:36:13: Custom config HEX string: 55818000002804a9313104b400
22:36:30: Settings 55 was send successful
22:36:30: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
22:36:30: Custom config HEX string: 55818000002804a9313104b400
22:36:35: Settings 55 was send successful
22:36:35: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
22:36:35: Custom config HEX string: 55818000002804a9313104b400
22:36:44: Disconnected.
22:36:46: Reconnect
22:36:47: Hardware Revision String: B1.6
22:36:47: Software Revision String: V4.6
22:36:47: Firmware Revision String: github.com/pvvx
22:36:47: Detected custom Firmware
22:36:47: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
22:36:47: Custom config HEX string: 55818000002804a9313104b4
22:36:54: Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/Z03MMC_v0117.bin'...
22:36:54: File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/Z03MMC_v0117.bin
22:36:54: File size: 143844 bytes
22:36:54: Count: 8991
22:36:55: Start DFU
22:36:56: Clear Ext.OTA region (141kib)...
22:38:10: Disconnected.
22:38:11: Reconnect
22:38:13: Hardware Revision String: B1.6
22:38:13: Software Revision String: V4.6
22:38:13: Firmware Revision String: github.com/pvvx
22:38:13: Detected custom Firmware
22:38:13: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
22:38:13: Custom config HEX string: 55818000002804a9313104b4
22:38:24: Settings 55 was send successful
22:38:24: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
22:38:24: Custom config HEX string: 55818000002804a9313104b400
22:38:33: Disconnected.
22:38:34: Reconnect
22:38:39: Hardware Revision String: B1.6
22:38:39: Software Revision String: V4.6
22:38:39: Firmware Revision String: github.com/pvvx
22:38:39: Detected custom Firmware
22:38:39: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
22:38:39: Custom config HEX string: 55818000002804a9313104b4
22:38:48: Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/Z03MMC_v0117.bin'...
22:38:48: File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/Z03MMC_v0117.bin
22:38:48: File size: 143844 bytes
22:38:48: Count: 8991
22:38:53: Start DFU
22:38:55: Clear Ext.OTA region (141kib)...
22:40:37: Disconnected.
22:41:09: Searching for devices
22:41:27: Connecting to: ATC_735252
22:41:29: Hardware Revision String: B1.6
22:41:29: Software Revision String: V4.6
22:41:29: Firmware Revision String: github.com/pvvx
22:41:29: Detected custom Firmware
22:41:29: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
22:41:29: Custom config HEX string: 55818000002804a9313104b4
22:41:48: Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/Z03MMC_v0117.bin'...
22:41:48: File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/Z03MMC_v0117.bin
22:41:48: File size: 143844 bytes
22:41:48: Count: 8991
22:41:53: Start DFU
22:41:54: Clear Ext.OTA region (141kib)...
22:45:09: Disconnected.
22:45:09: Reconnect
22:46:24: Searching for devices
22:46:25: Searching for devices
```

pvvx, 2023-12-15:
> What BT adapter are you using?

Windows:
On the main computer: AMD Wi-Fi 6E RZ616 (MT7922A22M) (Bluetooth v5.3 - HCI 12.5664/ LMP 12.8962)
Over: Intel® Wi-Fi 6E AX210 (Bluetooth v5.3)
USB-BT:
For testing - several options with CSR8510A10 (BTv4.2) chips, RTL8761BU (BTv5.3).
Bad adapters: ATS2851 (some of them cannot be used - there are a lot of errors in the firmware), some other unknowns and crafts from aliexpress that I haven’t thrown out yet...

pvvx, 2023-12-16:
I can only support some of Apple's recommendations. But this will be in the next firmware version, if the rest of the tests are successful.

MikealSouth, 2023-12-16:
Hey pvvx, thank you very much for the detailed explanation. I was about to order a BT USB dongle with the Chips you mentioned, when I realized that my Intel NUC desktop also has BT, using the AX201 chipset.
Everything worked like clockwork on the first try.
Then I set Z2M to "pairing", or whatever it is called, and a couple of seconds later the thermometer showed up!

Thank you very much for all of your efforts.

Log
```
9:42:15 AM: Searching for devices
9:42:19 AM: NotFoundError: User cancelled the requestDevice() chooser.
9:42:19 AM: Reconnect 1 from 5
9:42:26 AM: Searching for devices
9:42:32 AM: Connecting to: ATC_735252
9:42:45 AM: Hardware Revision String: B1.6
9:42:45 AM: Software Revision String: V4.6
9:42:45 AM: Firmware Revision String: github.com/pvvx
9:42:45 AM: Detected custom Firmware
9:42:45 AM: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
9:42:45 AM: Custom config HEX string: 55810000002804a9313104b4
9:42:50 AM: New custom config: [129, 128, 0, 0, 40, 4, 169, 49, 49, 4, 180]
9:42:50 AM: Settings 55818000002804A9313104B4 was send successful
9:42:51 AM: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.6, Sensor: SHT4x
9:42:51 AM: Custom config HEX string: 55818000002804a9313104b400
9:42:57 AM: Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/Z03MMC_v0117.bin'...
9:42:57 AM: File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/Z03MMC_v0117.bin
9:42:57 AM: File size: 143844 bytes
9:42:57 AM: Count: 8991
9:42:59 AM: Start DFU
9:43:00 AM: Clear Ext.OTA region (141kib)...
9:43:10 AM: Clean Flash OTA region: 0x040000, size: 144 kbytes. Go OTA...
9:44:41 AM: Update done after 90.268 seconds
9:44:45 AM: Disconnected.
```

closed 2023-12-17 by pvvx