pvvx/ATC_MiThermometer #442: Strange readings since 4.6 update

Toniob, 2023-12-17:
Hello,

I’m having strange readings since the last update.
[image]
As you can see, sometimes, the value sent by the sensor is wrong. These are the spikes on the graph. Strangely, the wrong value is always the same and doesn’t change.
[image]
Here, you can see the moment I updated the sensor and its behavior after that.

Toniob, 2023-12-17:
I have the same behavior on several sensors (they all are LYWSD03MMC 1.6). I’ve updated them all. 2 out of 6 do not have this behavior. I may be wrong, but those two were updated from 4.5 version, where the others with the issue were running the 4.3 version.

pvvx, 2023-12-17:
These are already known problems associated with the integration of BTHome into Home Assistant.

pvvx, 2023-12-17:
https://github.com/pvvx/ATC_MiThermometer/issues/418
https://github.com/pvvx/ATC_MiThermometer/issues/405

Toniob, 2023-12-19:
That’s ok, thanks. I’ve deleted the sensors from homeassistant, put them back, and it’s ok now.

closed 2023-12-19 by Toniob