pvvx/ATC_MiThermometer #444: advertisement message is too long (> 256 bytes)

bam80, 2023-12-22:
This probably a 3rd party problem but still I hope we could address it here.

My MQTT client app seems have a restriction of 256 bytes for the message length:
https://github.com/RadioShuttle/MQTTPushClient_Android

So I can't parse JSON's in the MQTT messages if they are bigger.

Could we have a setting to exclude some data from the advertisement messages, as temperature in Fahrenheit and such?
That would also add to the power-saving, maybe.

pvvx, 2023-12-22:
The thermometer uses messages no longer than 31 bytes, including headers. Bluetooth version 4.2 is no longer capable.
Bluetooth version 5.0+ can use advertising messages that are several kilobytes long.

bam80, 2023-12-22:
I see, but probably it transfers the temperature info twice - in Celsius and Fahrenheit?

pvvx, 2023-12-22:
No. Only in Celsius.

bam80, 2023-12-22:
Hmm, that means the Fahrenheit temperature is added by [Theengs gateway](https://gateway.theengs.io/)?

I get following MQTT JSON's:
```json
{"name": "ATC_*",
"id": "AA:...",
"rssi": -63,
"brand": "Xiaomi",
"model": "TH Sensor",
"model_id": "LYWSD03MMC/MJWSD05MMC_PVVX",
"type": "THB",
"tempc": 21.73,
"tempf": 71.114,
"hum": 34.24,
"batt": 100,
"volt": 3.074,
"mac": "AA:..."}

pvvx, 2023-12-22:
> means the Fahrenheit temperature is added by [Theengs gateway](https://gateway.theengs.io/)?

Yes. The gateway recalculates and adds

bam80, 2023-12-22:
Do you know if I could have control over the data included in JSON?

Other sample: `model_id` is pretty long and adds a lot to the message.

bam80, 2023-12-22:
> Yes. The gateway recalculates and adds

Do you know where it happens?

I only see it uses `tempc`:
https://github.com/theengs/decoder/blob/development/src/devices/LYWSD03MMC_json.h#L12

closed 2023-12-23 by pvvx