pvvx/ATC_MiThermometer #445: Cannot reconnect after flashing version 4.6

Geekay1972, 2023-12-22:
Today I connected my brand new LYWSD03MMC, it showed all values correctly and I flashed version 4.6 of the firmware.
After that it doesn't reconnect any more. I tried removing the battery but that didn't help either. Display is still working,

Here is the log:
[image]

mr-sneezy, 2023-12-22:
FWIW I have the same result from flashing to the Zigbee FW from BLE FW 4.6 with an older sensor. 
I can after many attempts get ZHA to detect and configure for the sensor in HA, but once past that HA receives no further sensor readings. Reset via the 3 second reset a number of times with same end result. Sensor LCD still displays OK. Battery was new when I changed to Zigbee FW. Not sure what to do with it as reverting the FW to BLE is not as easy as going the other way it seems.

pvvx, 2023-12-23:
Reset browser cache, BT adapter and drivers. In some embodiments, this is done by turning the power off and on.
The thermometer firmware has nothing to do with this.

closed 2023-12-23 by pvvx