pvvx/ATC_MiThermometer #446: Reverting from the Zigbee FW back to BLE FW ?

mr-sneezy, 2023-12-23:
I've had less than good results from flashing an older LYWSD03MMC to the zigbee FW via telink flasher from BLE FW 4.6. 
At this point I'd like to revert back to the BLE FW as I solid results with that. The LYWSD03MMC is not reporting to HA via ZHA 'except' for once during or right after binding to ZHA (then nothing further from it), so I doubt that I can revert via ZHA OTA.

Can somebody point me to a good place to read up on flashing the LYWSD03MMC via UART, as I have not issues with soldering to the board with my microscope ?

pvvx, 2023-12-23:
[The USB-COM adapter writes the firmware in explorer. Web version.](https://github.com/pvvx/ATC_MiThermometer#the-usb-com-adapter-writes-the-firmware-in-explorer-web-version)

mr-sneezy, 2023-12-23:
Thanks for the link, I had not found that one via Google so much appreciated. 
I have success. Back to BLE FW 4.6. I needed to connect reset line and use battery power before I got the success on this one (HW B1.4).
Thanks, Martin

closed 2023-12-23 by pvvx