pvvx/ATC_MiThermometer #447: cannot do activate after flashing orignial 026 MJWSD05MMC HW:U2.3'

wsctomy, 2023-12-23:
only shown known device id : blt.3.129vIdY9htg00 , others are blank 

5:31:23 PM: Searching for devices
5:31:28 PM: NotFoundError: User cancelled the requestDevice() chooser.
5:31:28 PM: Reconnect 1 from 5
5:31:45 PM: Searching for devices
5:31:47 PM: Connecting to: MJWSD05MMC
5:31:52 PM: Hardware Revision String: U2.3
5:31:52 PM: Software Revision String: 0026
5:31:52 PM: Firmware Revision String: 0026
5:31:52 PM: Detected Mi device
5:31:52 PM: Connected
5:31:53 PM: Activating now, please wait...

wsctomy, 2023-12-23:
solved by pressing the lower button

closed 2023-12-23 by wsctomy

pvvx, 2023-12-23:
This is described in detail in the instructions supplied with the device.