pvvx/ATC_MiThermometer #450: 10sec max is too low for Advertising interval

bam80, 2023-12-24:
MQTT app I use has limited amount of the packets data stored for graph's (about 100), so I need bigger Advertising intervals to have meaningful history deep.
And I don't need the data so frequently, anyway.

Is it possible to have wider Advertising intervals?

pvvx, 2023-12-24:
The maximum interval is limited by the Bluetooth standard version 4.2 to 10 seconds.

bam80, 2023-12-24:
In the other firmware, I see it can be set longer:

> Advertising interval
byte0 0xFE
byte1 0x06 - value times 10 seconds = interval 60 seconds default.

https://github.com/atc1441/ATC_MiThermometer#advertising-interval

pvvx, 2023-12-24:
This is not a "Advertising Interval".

bam80, 2023-12-24:
I bet I pasted it correctly, so your point probably needs further explanation.

Still, the other firmware allows to set the "Advertising interval" bigger?
Could we do the same here, as it might be really difficult to find a workaround.

pvvx, 2023-12-24:
https://github.com/atc1441/ATC_MiThermometer/blob/master/ATC_Thermometer/app_config.h#L9
3000 * 625 us = 1875000 us = 1875 ms = 1.875 sec.
https://github.com/atc1441/ATC_MiThermometer/blob/master/ATC_Thermometer/ble.c#L63C22-L63C42

bam80, 2023-12-24:
> This is not a ["Advertising Interval"](https://www.google.com/search?q=%22Advertising+Interval%22+max).

What is it then?
Can we have the same setting here, still?

pvvx, 2023-12-24:
> What is it then?

Read up on how Bluetooth works.

---

TelinkMiFlasher.html allows you to set the maximum time for transmitting new measurements to 250 seconds.

bam80, 2023-12-24:
I see it's not documented, as well as what that "new measurements interval" is and how it relates to the advertisement.
Probably worth to make a note in the Readme.

Anyway, can we make that other interval bigger, or it also restricted by some BLE spec?

bam80, 2023-12-24:
In the other fw I see it allows to set up to 10min of the "Advertising interval" (it's still not clear how it relates to BLE Advertising interval and to a "new measurements interval" here):
[image]
https://atc1441.github.io/TelinkFlasher.html

So we probably could make the our interval bigger than 250 sec, too?

pvvx, 2023-12-24:
Use firmware from atc1441.

bam80, 2023-12-25:
> TelinkMiFlasher.html allows you to set the maximum time for transmitting new measurements to 250 seconds.

As the interval for "new measurements" is not documented, could you clarify what it does exactly?

I had a strange behavior when tinkering with it.
If I use default values:
- 2,5 sec advertisement interval
- 2,5x4=10 sec of the new measurements interval

then the behavior is as expected:
I see new MQTT packets in the broker every 10 seconds.

But if I change the new measurements interval, things get wild and I start to receive the MQTT packets every advertisement interval or so.

pvvx, 2023-12-25:
This is the action of your MQTT broker. Contact its creator.

bam80, 2023-12-25:
OK, that might be.
Still,
> As the interval for "new measurements" is not documented, could you clarify what it does exactly?

pvvx, 2023-12-25:
“new measurements” is the transfer of new measurement results.

bam80, 2023-12-25:
>  The transmission packet contains a measurement number to help filter out duplicates.

It seems like [Theengs gateway](https://gateway.theengs.io/) I use just doesn't support it.
I'll file an issue there to suggest that feature.
Thanks.

bam80, 2023-12-25:
> I'll file an issue there to suggest that feature.

https://github.com/theengs/gateway/issues/202

pvvx, 2023-12-25:
Your request is not clear. What is the problem of transmitting each received BLE advertising packet by the gateway further to MQTT?
The function of the gateway is to transmit. The processing must be done on the end side, for example in the Home Assistant integrations themselves.

A constant reception together with duplicates indicates that the sensor is working and allows you to estimate the number of reception failures. That is, this is additional information and is needed in many cases.

It would be more correct if they send you away with such requests.

pvvx, 2023-12-25:
> MQTT app I use has limited ...

Write there.

bam80, 2023-12-25:
The feature I requesting is optional.

> It would be more correct if they send you away with such requests.

Thank you for the wish. All the best you too.

pvvx, 2023-12-25:
> The feature I requesting is optional.

Optional features lead to a lot of confusion for other users. And it takes time to parse, instead of doing what is needed, it takes time away from developers.

So you are engaged in sabotage?

closed 2023-12-25 by pvvx