pvvx/ATC_MiThermometer #451: LEScanIntervalAutoConnect, LEScanWindowAutoConnect values are unclear

bam80, 2023-12-24:
https://github.com/pvvx/ATC_MiThermometer/blob/ad46f4d36cbf3417829b39cdbffd95320cba35ae/README.md?plain=1#L88
Could you clarify how/when should I change them exactly?

Should they both correspond to a new "large" advertising intervals?
When the change is needed? What are the "large" advertising intervals?

pvvx, 2023-12-24:
Your questions are not related to the topic of BLE thermometers.
Do not duplicate questions - these questions have already been discussed.

---

Linux does not fully support the Bluetooth v4.2 standard, which was released back in 2014.
Each new version of Bluez has different fixes and new bugs dating back to 2014. Internal variables are constantly changing from version to version.
You can find out more about what and how to do to support BLE by posting questions on the relevant resources involved in creating the Linux kernel and Bluez.

closed 2023-12-24 by pvvx

bam80, 2023-12-24:
>  Do not duplicate questions - these questions have already been discussed.

Sorry I didn't face it. Where it has been discussed?

> Use a modern version of Android and there will be no questions. New versions of Bluetooth are supported there.

I have Android 8.0.0.
Unfortunately, I can't connect from there either..

pvvx, 2023-12-24:
Android 8 – 2017. Supports Bluetooth 5.0.
You probably installed an old chip, version 4.0.

bam80, 2023-12-24:
>  Do not duplicate questions - these questions have already been discussed.

Could you link the duplicated issue?
Maybe I'll find the info there

pvvx, 2023-12-24:
The beginning about Intervals is somewhere there
https://github.com/pvvx/ATC_MiThermometer/issues/172#issuecomment-1008288148